openyou/emokit #141: Python code does not work on Ubuntu 14.x

justinIs, 2015-01-26:
Here's my output:

```
Linux detected.
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/gevent/greenlet.py", line 327, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/justin/Documents/fydp/emokit/python/emokit/emotiv.py", line 404, in setup
    self.setup_posix()
  File "/home/justin/Documents/fydp/emokit/python/emokit/emotiv.py", line 472, in setup_posix
    serial, hidraw_filename = get_linux_setup()
  File "/home/justin/Documents/fydp/emokit/python/emokit/emotiv.py", line 214, in get_linux_setup
    for filename in os.listdir("/sys/class/hidraw"):
OSError: [Errno 2] No such file or directory: '/sys/class/hidraw'
<Greenlet at 0xb70fa7acL: <bound method Emotiv.setup of <emokit.emotiv.Emotiv object at 0xb70f6fac>>> failed with OSError

This operation would block forever
Traceback (most recent call last):
  File "example.py", line 17, in <module>
    print packet.gyro_x, packet.gyro_y
AttributeError: 'NoneType' object has no attribute 'gyro_x'
```

I am assuming that Ubuntu 14 does not support the hidapi stuff. Since I have nothing when i try:
`ls -l /dev/hidraw*`
Are there any alternatives?

bschumacher, 2015-01-26:
Your kernel is likely missing hidraw drivers, you could recompile it with them selected.

Thanks,
Bill Schumacher

justinIs, 2015-01-27:
Thanks for your response Bill. Can you give me a little more information on that please? I've tried installing ubuntu 12.04 here: http://releases.ubuntu.com/12.04/ and it seems that I may still be missing the drivers.

Is there any way I can install the drivers at this point? Or does it have to be compiled with the kernel - meaning that I have to install ubuntu again?

If the latter then can you perhaps point me to a distro with that?

Sorry for asking so much but my google searches aren't very helpful. Like me searching "ubuntu 12 with hidraw" doesn't yield what I really want.

Thanks.

fractalcat, 2015-01-27:
Are you running a 12.x release or a 14.x release? Your initial message implies 14, but later ones imply 12. There's no hidraw package available for 12.04; you could either upgrade to a later Ubuntu release (I suggest 14.04) and install libhidapi-hidraw0, or try and compile/install yourself.

bschumacher, 2015-01-28:
I’ll work on a write up later tonight.

Thanks,
Bill Schumacher

justinIs, 2015-01-28:
@fractalcat I believe I actually had everything working on 12.x last year. I did a fresh install of 14.x this year and had the exact issue that I described in my first post. So I tried with 12.04 after making that post.

@bschumacher thank you so much!

bschumacher, 2015-02-03:
Something changed in the way ubuntu 14.04 handles hidraw devices, the driver is still included in the kernel.

The emotiv dongle is still registered as hidraw1 and hidraw2 on my VM and works however it is not listed in the /sys/class/hidraw directories anymore.

Which effectively broke the detection, so now we need to come up with a new way of detecting it for linux devices. 

Thanks,
Bill Schumacher

fnielsen, 2015-12-10:
Is this fixed? I experienced no problem with a newly installed vanilla Ubuntu 14.04 and install from the emokit repo. I see devices appearing in `/sys/class/hidraw` when the dongle is plugged in:

    $ ls -al /sys/class/hidraw/
    ...
    lrwxrwxrwx  1 root root 0 Dec 10 14:26 hidraw1 -> ../../devices/pci0000:00/0000:00:1d.0/usb5/5-1/5-1:1.0/0003:1234:ED02.0006/hidraw/hidraw1
    lrwxrwxrwx  1 root root 0 Dec 10 14:26 hidraw2 -> ../../devices/pci0000:00/0000:00:1d.0/usb5/5-1/5-1:1.1/0003:1234:ED02.0007/hidraw/hidraw2

bschumacher, 2015-12-20:
My issue was related to VMware... I've noticed auto kernel module loading breaks it.

So if you're using a virtual machine don't enable that option when installing VMware tools.

I've not tested it with 14.04 without that, so maybe?

I'm installing a copy of 15.10 and will try with 14.04 later.

Today is dedicated to closing some of these issues, making the library more user friendly and to use the same requirements across all platforms.

So far it's going pretty good, only the windows is left.

bschumacher, 2015-12-21:
In my latest pull request #154 I've changed the library to use hidapi and pyhidapi, so it shouldn't matter if the hidraw files are there or not. Since by default hidapi uses libusb, although it can be changed to libhidraw when compiling hidapi.

Of course if the files are not there you might want to check dmesg and make sure your machine is detecting the dongle being plugged in.

sat-turn, 2016-03-28:
Hi im getting the same initial error to you, did you ever resolve it?

bschumacher, 2016-06-20:
Should work now. Pull the latest version, open a new issue if it persists.

closed 2016-06-20 by bschumacher