openyou/emokit #142: Python - Multiple Device Id's found

warrenarea, 2015-02-25:
I've been using emokit for quite a while now, and i've noticed my device has two
device id's. 

Epoc BCI
and
00000000000

setup loops through all hid devices and opens all matching
emotiv devices and because its multithreaded(simulated) it creates two
processes. this was causing my update_console to freeze up.
when i eliminated the epoc bci from connecting, it seemed to speed
things up.

my past usb dongle would connect as "Emotiv RAW DATA"  and only had that one entry
which makes me think others only have one entry as well.

so my question is, should the code be updated to connect only one device?
it appears the 00000000000 did take precedence.

I tried without the 00000000000 and only connecting to Epoc BCI and
the data being updated was empty.

Razorblade2k, 2015-04-29:
I've just bought an EPOC+ with raw EEG data option. I'm on Windows, and the device shows as 2 different entries (with the same PID/VID):

Brain Computer Interface USB Receiver/Dongle
and
EEG Signals

Similarly to what you mention, the first one returns empty data, and 'EEG Signals' is the useful one.

warrenarea, 2015-04-29:
can you post all your vendor id's it found?

also... are you certain this is an EPOC+ and not a regular Epoc ?
maybe also could you post a screenshot here showing your EEG signal data while wearing it for 20 seconds  (avoid wiggling)

thanks!

bschumacher, 2015-04-29:
Yeah the device checking part of the code is terrible, it was a quick fix and not a very good one.
On a mac it won't even attempt the next set of pid/vids it just crashes. I meant to fix this some time ago.
A timeout if no data is received might be a solution, although sometimes I have to flip the switch on my headset before I'll receive anything.

warrenarea, 2015-04-29:
i discovered my issue when displaying the greenlet threads. i think i remember having a headache trying to manually close them, the idea was to close all the greenlets to start over fresh on something like a time-out, but I eventually gave up on that option as it kept crashing. I really think gevent is long over-due for an overhaul. I tried to port it to py3 but i eventually hit a wall. makes me wonder if we wouldnt be better off without the gevent/greenlet setup entirely.

bschumacher, 2015-04-29:
I'll give it a try and see if I can make it work, gevent recently added py3 support I believe as well.
There might be another solution as well but without gevent the performance is pretty bad, from what I remember.

bschumacher, 2015-12-21:
This should be resolved now, with my latest pull request. #154 

Sorry it took so long.

warrenarea, 2015-12-23:
very good, i look forward to trying to set up a port for python3,
last i remember it was only a bunch of cosmetic print/error changes.
i'll have to add it to my list of projects though.

warrenarea, 2015-12-24:
https://github.com/CymatiCorp/CyKit-Python-3.3
ported cykit for Python 3.3.0 for use with the new gevent port.

i havn't thoroughly tested it because my battery seems to
have been depleted from inactivity. However, i got it to the point
where it operates as if it should work. If that makes any sense.

there wasn't really much involved, but as i said, some trivial
cosmetic print/exception changes.

The only hindrance is that it now requires VC++2010 to operate
and install gevent.

warrenarea, 2015-12-24:
i can now confirm that the Python 3.3.0 port is connecting and working, 
i will have to do some tinkering to sort out some data display issues,
but at least we have a working prototype.

bschumacher, 2015-12-24:
Very nice.

warrenarea, 2016-01-05:
https://github.com/CymatiCorp/CyKit-Python-3.3

Python 3.3 is now completely ported and working, took me several days
before finally sorting out the correct unicode format that worked with
cipher.decrypt() 

also set it up to ask the user what device they want to use.

warrenarea, 2016-01-20:
from what I've seen, there isn't any obvious advantage to using Python 3.3
over 2.7, other than using the latest distributions. 

you won't see any noticeable increase in speeds, and the major difference
in python 3 that i can see, is that you have to be a lot more specific with the
text encodings.  I think what slowed me down the most was that I kept 
overlooking the "latin-1" encoding set, because I thought it was for another
language entirely and not applicable.  However it was actually the encoding
set i needed to convert to the full ASCII charset.

nathanielatom, 2016-03-02:
I had a similar issue with multiple usb devices showing up on linux, with Epoc 2013. For example, I would get hidraw1 and hidraw3 both with Emotiv as the manufacturer, but data would only stream to hidraw3. I created a pull request #158 which I believe solves this issue for linux.

warrenarea, 2016-03-02:
i think my latest CyKit version deals with this, by displaying all of the device handles, and then
asking the user to choose which one they want to use by entering a number.

can be a little tedious loading it every time, but its the best one-size-fits-all solution i can come up
with at the moment.

nathanielatom, 2016-03-02:
I checked out CyKit, but it looked like only windows was supported. I'm running xubuntu on a Pi.

warrenarea, 2016-03-02:
yes, you are correct... i was trying to strip it down to its bare minimum to make it easier
to understand how it worked at the time. 

plus i have a port now that utilizes the latest gevent and python 3.3, which has very little
obvious advantage mind you. though not having a linux os setup it would have been a bother
to attempt to port.

bschumacher, 2016-06-20:
This is corrected with the latest version.

closed 2016-06-20 by bschumacher