openyou/emokit #143: EEG option needed?

JanBam, 2015-03-24:
This is rather a question than an issue, but I've found no answer in the FAQ.

When buying an Epoc there's an EEG option ($300). The store claims, that this option includes a different firmware for getting raw EEG data.

Do I need this option for using emokit?

Razorblade2k, 2015-04-29:
I'm new here and I don't know the actual answer, but after having paid +$300 for the "raw EEG data" option, it seems to me that emokit can read raw data even without that option.

Emotiv says that the difference is in the firmware. jamesbeedy said in another thread that the difference is just the SDK software. I'm confused.

llSourcell, 2015-09-01:
im curious of this as well. Anyone have an answer? @jamesbeedy

nikhiljay, 2016-03-30:
Check this out: https://github.com/openyou/emokit/issues/139

bschumacher, 2016-06-20:
Raw data is available using emokit.

closed 2016-06-20 by bschumacher