openyou/emokit #144: Can i connect non-eeg headset to eeg- usb dogle?

mnzab, 2015-04-16:
I purchased a research edition SDk of Emotiv neuroheadet  (eeg ) and after  a couple of years the headset is not in usable condition. Recently i have purchashed a non-eeg system. Can this non-eeg headset work with the eeg-sdk? I have tried to pair up the non-eeg headset with the eeg-usb dongle but was not successful in that. Can anyone help me getting a solution.

jamesbeedy, 2015-04-16:
@Mnzab no.

jamesbeedy, 2015-04-16:
@mnzab JUst joking. Why do you think you would be able to do that? What do you mean non eeg?

jamesbeedy, 2015-04-16:
@mnzab Do you mean you have another device that transmits/receives 2.4ghz and you want to use the emotiv wireless dongle to communicate with that device? If so, that is highly possible, depending on your other device .... might take some hacking for sure.

mnzab, 2015-04-16:
@jamesbeedy, Thanks .  Once we place a purchase order on emotiv website, it asks for the option (raw eeg data Yes/NO) . If we select yes, they term that system as eeg, else non-eeg.  I asked this because i learnt from forums that in emotiv systems,  the  encryption is implemented in the USB dongles and not in the headsets. If this was the case then the firmware for all headsets (eeg or non-eeg) should be the same.

jamesbeedy, 2015-04-16:
@mnzab Ahh I see. Both physical components are the same, the only difference is you get the sdk with the research package.

jamesbeedy, 2015-04-16:
*raw data package

mnzab, 2015-04-16:
I have that package with me along with its USB dongle but its headset  had gone  bad. I purchased another system last month but  this is non eeg . Now i want to use this headset instead of the older one.  What is understand from your comments that headsets , whether they come with the eeg system or non-eeg system , are the same and can be used interchangeably. Is this right?

jamesbeedy, 2015-04-16:
Yes. Concerning terminology...both devices are eeg devices. The difference is one comes with an sdk.

jamesbeedy, 2015-04-16:
@mnzab if your headset is broken you should send in for repair.

Razorblade2k, 2015-04-29:
I would like to add that it seems there are 3 different dongles (same price):

- Emotiv USB Receiver Universal model:
Replacement USB receiver for Emotiv EPOC, EPOC+, and Insight Neuroheadsets.

- EPOC USB Receiver (EEG):
Replacement USB receiver for the Emotiv EPOC neuroheadset with EEG access.  

- EPOC USB Receiver (non-EEG)
Replacement USB receiver for Emotiv EPOC Neuroheadset with no EEG access.

If the photos on their webpage are correct, I have received a "Universal model" dongle with my new EPOC+ plus "raw EEG data" option.

closed 2016-06-20 by bschumacher