openyou/emokit #145: EPOC+ limited licence issue

tedor2, 2015-05-22:
Hi Cody, 

The university's EPOC+ allows us to raw data in Emotiv's Testbench app, however has limited licenses (1 MAC + 1 PC). I am interested in doing DSP on a 3rd computer with SuperCollider (OSX), which I believe I could use Emokit for. 

Just double checking:
Will using the Emokit with the USB dongle change the dongle settings in any way? 

Thanks, 

Best wishes, 
Kris

Razorblade2k, 2015-06-02:
AFAIK, it won't change anything; you can use Emokit with no risk (Emokit does not write any configuration).
But if you want it to work with EPOC+ (plus), you have to adapt the code as I described in this thread: https://github.com/openyou/emokit/issues/138

Regards.

tedor2, 2015-07-09:
I think I will have to buy my own headset to experiment with Emokit as I don't dare messing up the licences/dongles of the university. 

Cheers.

fractalcat, 2015-07-09:
Hi,

Emokit will not change the dongle settings in any way (that I've been able to detect or find remotely plausible); the interface is read-only.

closed 2015-07-09 by fractalcat