openyou/emokit #146: is the emotiv insight supported?

beckmx, 2015-07-14:
Hello guys, does anyone know if this library supports the emotiv insight?

danielw93, 2015-07-14:
TL;DR
No I'm afraid it isn't and also likely can't be easily adapted.

In the current development status the Emotiv Insight can't possibly be supported by the emokit library for mainly 2 reasons:
1. The original decryption that is used in the emokit library was reverse engineered for the first version of the Emotiv Epoc back in late 2011 and is specific to exactly that group of devices. Because a hardcoded key that is embedded on the USB dongle receiver is used to decrypt the data send by the headset. And it is very likely that they changed the key for the Insight/used a different system altogether.
(Example: just look at the problems that arise when trying to run emokit with the newer version, the EPOC+ #138 )

2. The script as it is right now, is programmed to interpret the incoming packets as information from 14+2 (reference) electrodes as well as 2 gyro axis.
While the quantization (0.51µV least significant bit) and data resolution (128 samples per second) seem to be the same for the old Epoc and the new Insight, the data encoded in these packages would need to be interpreted very differently from each other because the Insight only has 5+2 (reference) channels.

This means  significant investigation would have to be done in order to figure out the new encryption scheme as well as figuring out which bits on the package carry information about which channel of the Insight.
And in the end you would most likely be ending up rewriting most of the https://github.com/openyou/emokit/blob/master/python/emokit/emotiv.py script in order to make it work.

fractalcat, 2015-07-14:
@danielw93 is correct; unfortunately emokit does not support the Insight, and I'd say that's unlikely to change (I don't have an Insight, and the library structure is such that supporting another device would require some significant changes).

closed 2015-07-14 by fractalcat