openyou/emokit #147: ECB mode does not use IV

vikram186, 2015-08-05:
I have an EPOC and was trying to use the emokit. After fixing the VID, PID and udev rules, it was by the example.py. After printing the Serial numbers there was no output. When I pressed "ctrl+c", it prints this backtrace and terminates. The "ECB mode does not ..." seems to come from the pycrypto library.  Any ideas would be much appreciated.

Here is the complete backtrace:
```bash
$ sudo python example.py 
Serial: SNXXXX Device: hidraw0
Serial: SNXXXX Device: hidraw1 (Active)
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/gevent/greenlet.py", line 327, in run
    result = self._run(*self.args, **self.kwargs)
  File "/opt/devel/courses/ss-15/Alight/emokit/python/emokit/emotiv.py", line 518, in setupCrypto
    cipher = AES.new(key, AES.MODE_ECB, iv)
  File "/usr/local/lib/python2.7/dist-packages/Crypto/Cipher/AES.py", line 180, in new
    return AESCipher(key, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/Crypto/Cipher/AES.py", line 113, in __init__
    blockalgo.BlockAlgo.__init__(self, _AESNI, key, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/Crypto/Cipher/blockalgo.py", line 398, in __init__
    self._cipher = factory.new(key, *args, **kwargs)
ValueError: ECB mode does not use IV
<Greenlet at 0x7f902c9e3cd0: <bound method Emotiv.setupCrypto of <emokit.emotiv.Emotiv object at 0x7f902c9a6f50>>('SNxxxxxxxxxx')> failed with ValueError

This operation would block forever
Traceback (most recent call last):
  File "example.py", line 11, in <module>
    print packet.gyroX, packet.gyroY
AttributeError: 'NoneType' object has no attribute 'gyroX'
```

bschumacher, 2015-08-09:
Those values are definitely wrong, you're not decrypting the values correctly. Probably due to the IV. What version of PyCrypto are you using?

vikram186, 2015-08-09:
I am using pycrypto (2.7a1). Since that is an experimental version, I have tested it with the stable version (2.6.1). I am getting the same "Value error: ECB mode does not use IV" if I use IV. If I take off the IV from the parameters, I am getting erroneous values.

bschumacher, 2015-08-10:
I'll have to due some research, perhaps someone else already knows.

vikram186, 2015-08-12:
Hi, I have played around with the 'type' value in the setupCrypto() and it works now.  I am getting correct battery values. The IV does NOT play any role now as mentioned in the commit of pycrypto [8bde0dc](https://github.com/dlitz/pycrypto/commit/8dbe0dc3eea5c689d4f76b37b93fe216cf1f00d4). This has happened in pycrypto repository after the release of v2.7a1. So, if someone uses the latest master of pycrypto instead of using pip install, they will encounter this error. Until there is a new release from pycrypto, I don't think this needs to be fixed at our end. Kindly let me know your opinion on this, then I can send a pull request. As of now, I am closing this issue. Thanks for your support @bschumacher

closed 2015-08-12 by vikram186