openyou/emokit #148: Data rate too low

ssaz5, 2015-08-18:
From what I know Epoc samples the data at 128Hz and with emokit I am getting a sampling rate less than 40Hz. Can you please tell me what could be wrong?

olorin, 2015-08-18:
Hi,

> From what I know Epoc samples the data at 128Hz and with emokit I am getting a sampling rate less than 40Hz. Can you please tell me what could be wrong?

Not without more data, no. Can you provide steps to reproduce?

-- 
OpenPGP: 6FB7 ED25 BFCF 3E22 72AE 6E8C 47D4 CE7F 6B9F DF57

danielw93, 2015-08-18:
Wait...do you mean the actual sampling rate or do you mean the bandwith of the device?

i.e. are you getting less than 128 packets per second from the device? 
(that should show a bunch of lost packets when you run the example.py program), or are the signals you're getting less than 40Hz in frequency? That would be normal then because the bandwith of the Emotiv Epoc headset is between 0.2Hz and 45Hz.

ssaz5, 2015-08-18:
The signals i am getting are less than 5Hz in frequency and i was getting a sampling rate of 40 samples/sec. Though the second issue is resolved now the first one isn't. 
While i am moving my body i should get some frequency content around 10 -20 Hz but all the sensors are giving a frequency less than 5 Hz.

ssaz5, 2015-08-18:
Turns out i was using emokit wrong. I am getting the data at 128 samples/sec now but when i perform fourier transform on the data the frequency content of the signal is almost zero and when i plot the signals in matlab i get almost fixed voltages with fluctuations everytime i do something. But the frequency of the signals is too less for eeg signals. 
Kindly tell me what i could be doing wrong. The epoc control panel is working just fine so i atleast know that my headset is fine. 

Hi,

Not without more data, no. Can you provide steps to reproduce?

bschumacher, 2016-06-20:
Without an example of output or code it is nearly impossible to help you, although it looks like the library is functioning correctly. If you want to open a new issue with some examples I would be more than happy to take a look. Closing for now.

closed 2016-06-20 by bschumacher