openyou/emokit #153: No Data Received on Python under Ubuntu

medo, 2015-12-16:
Whenever I execute example.py I got the following.

```
Linux detected.
Serial: sN20131213002720 Device: hidraw0
Serial: sN20131213002720 Device: hidraw1 (Active)
```

No data is received after that point.

bschumacher, 2015-12-16:
> <reply+0022e3b9017e98d2aa6749622095e452898d0b77e5216ed992cf000000011288894d9
> 2a169ce074bd840@reply.github.com>

What version is your Ubuntu and kernel?

<https://github.com/openyou/emokit/issues/153> .

medo, 2015-12-16:
Ubuntu 12.04 LTS
Linux precise32 3.2.0-23-generic-pae

bschumacher, 2015-12-16:
> <reply+0022e3b93e2778632d1faa043671d01ab4365aef3fe260eb92cf0000000112889af09

I¹m having issues as well, I¹ll see if I can fix it. Probably not going to
be until later though.

Date:  Tuesday, December 15, 2015 at 9:22 PM
Cc:  William Schumacher <bill@servernet.co>
Subject:  Re: [emokit] No Data Received on Python under Ubuntu (#153)

<https://github.com/openyou/emokit/issues/153#issuecomment-164977246> .

bschumacher, 2015-12-16:
You might try toggling the power switch a couple times or move the headset
closer to the dongle and toggle the power.

My issue was with VMware fusion it seems, the light on the dongle should
start blinking rapidly if it is not then no data is being transmitted.

Please verify the light that normally blinks slowly when no headset is
connected does start blinking rapidly and still no output is produced.

<https://github.com/openyou/emokit/issues/153#issuecomment-164977246> .

medo, 2015-12-16:
The light is blinking correctly but no output is produced. I am running the linux image under VirtualBox.

bschumacher, 2015-12-16:
> <reply+0022e3b9b5f13eb508a14b5b3a7b64f9e0f2ce12381669e392cf00000001128948d49

How long ago did you order your headset? What model is it?

Date:  Wednesday, December 16, 2015 at 9:44 AM

<https://github.com/openyou/emokit/issues/153#issuecomment-165149620> .

warrenarea, 2015-12-17:
i noticed that it found two HID usb handles, perhaps you could try
manually selecting the first one (0), and see what the results are.

It does sound like the device is working.  when it blinks rapidly, the
led light also dims as well.

not entirely relevant, but i think i had it detecting two HID's on mine 
as well, and it was creating two threads and updating the information
at the same time, producing a noticeable lag.  so selecting the HID
manually is probably a preferred solution anyways.

though from the sound of it, it could also do with how ubuntu works with
usb devices.

warrenarea, 2015-12-17:
you might also try printing out your vendor.id and see what that says.

warrenarea, 2015-12-17:
i'm not sure if they've updated it, but this might prove relevant.
https://github.com/openyou/emokit/issues/94

at the bottom, they suggest there is a naming change.

bschumacher, 2015-12-20:
Make sure you are using sudo as well.

warrenarea, 2015-12-24:
if Ubuntu has an EmotivControlPanel application, perhaps you could try running that and see
if its pairing up properly and sending data to that. worth a try.

nathanielatom, 2016-03-17:
I had a similar issue on ubuntu 14.04 and xubuntu 14.04 where 2 usb devices show up for the Epoc. They are not always contiguous (one could correspond with hidraw3, and the other hidraw5) but as far as I could tell, the latter always was the one that contained real data (the former just pulling zeros). I believe this pull request #158 fixes the issue.

bschumacher, 2016-06-20:
Latest version should fix your problem.

closed 2016-06-20 by bschumacher