openyou/emokit #155: Dongle not pairing correctly.

warrenarea, 2015-12-24:
I had an interesting issue today,

I hadn't used my modified headset for several months
and I was pretty certain I had it switched off, however 
the battery seemed to be dead.  So I plugged it in for
a few hours and try to pair it... with the headset almost
directly on top of the dongle. 

I repeated this process several times, would let it charge
a while, and then plugged in the dongle several times. 
while turning the headset on/off. 

I tried different USB ports, updating usb ports, letting it
reinstall the drivers... although which was kind of odd
was that when Windows was auto-detecting the device
drivers, at first the Device names appeared "Emotiv Device"
or what have you. then once configured they were changed
to "USB input device"... however the hidapi was still detecting
the names... thats odd, but not what i thought was interesting.

no matter what I did, one green light kept blinking green.
i even hydrated the sensors, thinking maybe it needed to 
synchronize the headset data first, but still no data light.

after it had charged for several hours, and I was certain
that it was no longer an issue with the battery, i had a
brainstorm, that maybe it will connect if I use Testbench
or EmotivControlPanel. 

Sure enough, a minute or two of opening/closing EmotivControlPanel 1.0
and turning on/off the device, the data light comes on. 
however, the data light was bright and solid green, (not blinking)

i tried it again with emoKit/Cykit and it worked for briefly, then stopped,
then it even stopped working for EmotivControlPanel 1.0

So I tried it with the latest EmotivControlPanel 2.0 sdk and moments
later the data light comes on and this time blinks rapidly and dimly, and works
as expected.

I realize this isn't an ideal scenario, since the battery could have been
a related issue, however I remember previous testing in the past, where
while using emoKit that it stops pairing, and only pairs up again while 
EmotivControlPanel or Testbench is running. 

So I'm wondering if maybe there isn't some additional communication going
on that is related to the pairing of the devices?

bschumacher, 2015-12-24:
Are you running Windows 10? I've had driver issues with a couple of devices on it.

I've never had the issue you are talking about on OS X, Ubuntu/Debian or Windows 7. I too had a dead battery when I worked on this the other day.

bschumacher, 2015-12-24:
Although, I don't even put the sensors in anymore because the headset is busted I usually just place the headset close to the dongle because it has extremely limited range.

warrenarea, 2015-12-24:
Windows 7, i don't think it was a driver issue, but it was odd that 
windows detected the device name to begin with, then changed it to something else.

warrenarea, 2015-12-24:
they are definitely not very durable, which was part of my motivation for
modifying mine to begin with, so that i could get a good stable connection.
the elastic headband i made for mine works very well for that.

warrenarea, 2015-12-25:
okay i solved the issue, and it was nothing related to the EmotivControlPanel 

apparently the USB stick was bent just slightly, which caused the 475A capacitor
to break lose, so that when you put pressure on the stick it would cause the 
capacitor to make contact, and function normally, pairing the device.

I went to solder mine back on, but the prong was too deeply embedded, and the
capacitor is almost small enough that you can't operate on it. 

I tried to remove the capacitor out of my spare dongle, but as luck would have it,
it was cemented in place with no evident solder points... Not really wanting to 
absolutely destroy a perfectly good dongle for one capacitor, I soldered a couple 
of wires to the capacitor, glued them for a more solid connection, and taped
the dongles together. 

an electrical engineers nightmare, but it works!

warrenarea, 2015-12-25:
for those of you curious about the internals of the dongle, 

theres a sticker inside each dongle with the SN, however its missing the last 3 characters.

the antenna used to pick up the wireless signal, is housed in the end of the dongle in a
hook-shape line of solder.

there is a microchip inside that looks like it could be easily jtagged if one was inclined to do so.

warrenarea, 2015-12-28:
Also considering that the dongle is just an antenna for the headset
and there does not appear to be a transmitter, i no longer believe there to be a 
back and forth communication between the headset/dongle, and is a
one-way communication bridge, from headset to dongle.

my original hypothesis that the emotivcontrolpanel was involved seemed
to be a coincidence brought about by my faulty hardware.

closed 2015-12-28 by warrenarea