openyou/emokit #160: No packets received but successful setup on OSX 10.8 using EPOC+

ejolly, 2016-03-10:
I can get everything setup and working with no errors, including dependencies, but whenever I run the test program on the README I see no packets being received. Here's my test script:

```
from emokit import emotiv
import gevent
import hid

if __name__ == "__main__":
	devices = hid.enumerate()
	emotivDevs = [dev for dev in devices if 'Dongle' in dev['product_string']]
	serial_number = emotivDevs[0]['serial_number']	
	headset = emotiv.Emotiv(serial_number=serial_number, is_research=True)
	gevent.spawn(headset.setup)
	gevent.sleep(0)
	try:
		while True:
			packet = headset.dequeue()
			print packet.gyro_x, packet.gyro_y
			gevent.sleep(0)
	except KeyboardInterrupt:
		headset.close()
	finally:
		headset.close()
```

I know I can get a connection using this dongle and headset using Emotiv's software, and I've checked all the obvious things with my setup (and other suggestions from different github issues), including turning the headset off and on, changing the is_research flag and changing the vendor and product id's (mine were the same as line 515 in the emotiv.py file, although reversed, but neither ordering working and hid never returns an error finding the device). Direct calls to hidDevice.read() seem to return empty arrays. 
Any suggestions about what I might be doing wrong?

[image]

nathanielatom, 2016-03-17:
Hi, I recently got emokit working on OS X 10.11. 

I hardcoded the VID and PID (in hex) in emotiv.py, using the output from the try.py file from https://github.com/gbishop/cython-hidapi. Note however that it returned an incorrect serial number. This allowed me to connect and pull "data" but the encryption didn't work (based on the serial number) and all I got was white noise. I found the serial number by plugging the Epoc dongle into another device (windows or linux OS shows the serial number when you plug in the dongle).

Hope this helps.

nikhiljay, 2016-03-31:
I am stuck... How did you hardcode the VID and PID in hex in emotiv.py?

nathanielatom, 2016-03-31:
Sorry you need to edit line 509 of emotiv.py (https://github.com/openyou/emokit/blob/master/python/emokit/emotiv.py#L509)

Do you get output from try.py? use `hex(vid)` and `hex(pid)` where `vid` and `pid` are the integers printed for Emotiv from try.py.

bschumacher, 2016-06-20:
You might try the latest version, although someone else reported the values are not correct for the EPOC+.

bschumacher, 2016-06-20:
If it is not working with the latest version, open a new issue. Comment in issue #166 if your data is not correct as I suspect it will not be. Closing this issue.

closed 2016-06-20 by bschumacher