openyou/emokit #161: Issue with running emotiv.py

sat-turn, 2016-03-28:
Hi, ive installed all of the required libraries and depencicies successfully. however i am unsure on how to basically run the python library.

ive run emotiv.py and here is the result, any sort of guidance would be great:D

sanchir@ubuntu:~/Downloads/emokit-master/python/build/lib.linux-x86_64-2.7/emokit$ python emotiv.py build
Linux detected.
Traceback (most recent call last):
  File "emotiv.py", line 635, in <module>
    a.setup()
  File "emotiv.py", line 402, in setup
    self.setup_posix()
  File "emotiv.py", line 470, in setup_posix
    serial, hidraw_filename = get_linux_setup()
  File "emotiv.py", line 215, in get_linux_setup
    real_path = check_output(["realpath", "/sys/class/hidraw/" + filename])
  File "/usr/lib/python2.7/subprocess.py", line 566, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

nikhiljay, 2016-03-28:
I have the same problem. Thanks for asking.

nikhiljay, 2016-03-28:
I am getting this error:
nikhils-mbp:emokit Nikhil$ python emotiv.py
Traceback (most recent call last):
  File "emotiv.py", line 9, in <module>
    import hid
ImportError: No module named hid

sat-turn, 2016-03-28:
@nikhiljay have you installed hidapi yet?

nikhiljay, 2016-03-30:
Ok I am getting your exact error. Have you resolve it yet?

sat-turn, 2016-03-30:
@nikhiljay havnt had any luck since the post, please let me know if you do resolve it. ill keep at it as well

nikhiljay, 2016-03-30:
Oh I think I got something. I changed my directory from the emokit directory to the python directory and then tried to run "python example.py". This was after I inputted the serial number and configured emotiv.py in the emokit directory. After running example.py, I got this output:

[image]

Later today, I am going to connect the emotiv and test it out to see if I get any actual values :)

nikhiljay, 2016-03-31:
GOT IT WORKING! :smiley: 

[image]

I have the instructions in the readme of my repo: https://github.com/nikhiljay/brainwaves

sat-turn, 2016-03-31:
@nikhiljay that brillaint!, thanks very much for the guide. i was going through it and in step 5 to 6 
5) A hidtest file should be created in the same directory. Open the hidtest to run it.

6) The output should be a list of devices that are connected to the computer. Look at the Emotiv device information and copy the Serial Number.

i cant seem to compile the cpp file, initially it said the an hidapi.h was missing but i pasted in from another directory then a series of errors came up.  how did you get past this?

[image]

bschumacher, 2016-06-20:
Looks like your kernel doesn't support hidraw, you might try pulling the latest version.

bschumacher, 2016-06-20:
Which version of ubuntu are you running? Did you compile a custom kernel?

bschumacher, 2016-09-06:
If you need further assistance please open a new issue, after re-reading this it appears your problem is related to the compilation and installation of the hidapi library.

closed 2016-09-06 by bschumacher