openyou/emokit #162: Values

nikhiljay, 2016-04-03:
Hi, I was able to get the raw data out of the Emotiv, but I am confused how to interpret them. Any clue to what these numbers mean?

[image]

warrenarea, 2016-05-31:
yes its .51µV (microvolts)
where epoc+ is more sensitive and able to pick up .13µV
how epoc+ is able to pick up on emotional states.

the values here can be in the range of  -8192 and 8192

with the headset taken off, each sensor value will fluctuate slightly
from its baseline reading. (ie, the number that is most consistent)

So sensor `F8` here is -150, when it receives an EEG signal, it will increase
slightly, and then go back to -150 (providing -150 is the baseline)

What I prefer to do, is to average out the sensor quality over about a dozen
or so data samples... and this will give you the baseline number, that you can
then subtract, so that it will display more appropriately a reading of 0 when 
no EEG signal is received.

the Quality, is to indicate if the sensors have a clean connection between the
gold electrode, the felt pad filled with saline, and the pores of your skin, which
slightly absorbs some of the saline (or liquid). 

The value is 0-4 and indicates the strength of the signal that is coming through.

In this case, it probably wasn't attached to your noggin.

warrenarea, 2016-05-31:
more info here:
https://github.com/CymatiCorp/CyKit/blob/master/FAQ.md

closed 2016-06-20 by bschumacher