openyou/emokit #165: Intermittent Dongle Connection Issue

warrenarea, 2016-05-31:
I've mentioned this issue in the past, and because the issue
corrects itself, makes it hard to recreate. 

Whenever I go without using the headset for a few months, and
get a decent charge on it, and turn it on with emokit.

It starts working for a handful of times.

Then i noticed it gets really intermittent and drops packets, then emokit
just stops receiving packets entirely.

The only way I can get it to work again, is to open up Emotiv Control Panel 2.0
ECP v1.0 doesnt seem to have any effect. 

The "ECP 2.0" switches rapidly between a few states,  
indicating that the battery meter isn't detected, then the wireless isn't detected,
then it receives a Status: OK.   it does that cycle a few times, and then after that,
emokit seems to work flawlessly again.

i did have a similar issue once, where apparently the USB had been slightly bent
and the plastic housing was rubbing up against a capacitor, causing it to lose its
connection... but i fixed that issue, and i've seen it do this kind of thing in the past
before.

its not really an issue for me, but perhaps if anyone else has a similar experience,
they might try running Emotiv Control Panel 2.0 and see if it resolves it for them.

bschumacher, 2016-06-20:
Adding this to a wiki page or file linked to the readme might be good.

bschumacher, 2016-06-20:
Made a wiki page, closing issue. Will make the wiki pretty later, added your name to the bottom.

closed 2016-06-20 by bschumacher

warrenarea, 2016-06-21:
https://github.com/CymatiCorp/CyKit/blob/master/FAQ.md

you are welcome to use the FAQ i put together for CyKit, 
most of the information is either taken directly from the emotiv website,
or from some forum articles from gmac. so you can be sure that most 
of it is pretty accurate.

you may have to make some modifications to the FAQ to take out a few
CyKit related questions, but shouldn't take but a minute, and has some
useful resources/questions.

warrenarea, 2016-06-21:
you're also welcome to use this graphic, if you want to.
http://cymaticorp.com/edu/emokit.png