openyou/emokit #166: Epoch vs Epoch+, processed packet

Gabrock94, 2016-06-16:
Hi guys,
I've tried the python tool with both an Epoch and an Epoch+, but i have some issue. 

This is the old Epoch:
[image]

While this is the Epoch plus.
[image]

Any idea about how to solve this?

nikhiljay, 2016-06-17:
Sorry, I am not sure how to fix your problem. However, I am curious how you got the "Quality" values working. Any help?

Gabrock94, 2016-06-17:
Acutally, they are not working. The headset was placed on a table, so those value are not good. 
I think the problem is in the the "bit mapping". Probably they changed something between the versions, but i don't have the original software so I can't figure out how to download raw data and compare them with those extracted by python in order to remap the whole system.

nikhiljay, 2016-06-17:
Ah, I see. Good luck!

bschumacher, 2016-06-20:
Seems like you would want to add raw dump to file, before processing the packet. Then iterate over the data with a test script.

Gabrock94, 2016-06-21:
Yes, I'm trying to createa CSV file with the raw data. Actually it seems to work at 128sample per second with the old emotiv, but at around 80 sample per second with the new one. 
I think the problem is in the bit mapping, but could also be something around the processing of the package with pycrypto. Maybe I can try to store the raw package and process them later, but I'm not sure this is a great way to solve the problem. 
I'll try the test script, Thanks :)

bschumacher, 2016-06-21:
I have exporting data on my todo list. I worked on it last night a bit. Seems kind of clunky but gets the job done. I'll commit what I have here in a couple hours or so.

bschumacher, 2016-06-21:
When you get a chance pull down the latest code, change write=True and post the .csv on pastebin or something.

Gabrock94, 2016-06-23:
Thanks again for your help.
This is the pastebin of the firsts rows of the dump, let me know if you need more rows / seconds.
http://pastebin.com/Z9ChPULB

Note that I had to comment the line " print packet.gyro_x, packet.gyro_y" in example_exporter in order to use it.

bschumacher, 2016-06-23:
What is your serial number?

Gabrock94, 2016-06-24:
3B9AD8BE

jspenc72, 2016-06-24:
Hey Just noticed the recent activity related to Epoc +. Anything I can do to help? Just bought an Epoc + headset but wondered where you are at with adding support for it?

bschumacher, 2016-06-24:
Are you sure that's the whole serial number @Gabrock94 ? Did you get that from the output of the latest export example script?

bschumacher, 2016-06-24:
@jspenc72 Sure, if you could do the same thing get some output captured with the headset sitting on your desk/table and post the serial number.

bschumacher, 2016-06-24:
It is interesting that there are 8 characters and there are 8 indexes in the key we replace.

Gabrock94, 2016-06-27:
@bschumacher nope, it was somewhere in their documentation. I'll try asap with the script and i'll let you know. Thank for the help.

Gabrock94, 2016-08-22:
Hi all.
i tried to use this fork `https://github.com/a455bcd9/emokit and changing the variable is_research to True  it returns some results. It seems to work at 128 sample per seconds, with just a few packet lost in 10 seconds . Also the channel mappins seems good: Control Panel was showing no signal from channel O1, and recorded data too. 

[image]

CSV file
[testplus.zip](https://github.com/openyou/emokit/files/429513/testplus.zip)

jspenc72, 2016-08-22:
Is the data you posted from an epoc+?

bschumacher, 2016-08-22:
Very nice, the data looks good. That is_research True does make a difference in the crypto key so it looks like they are using that now.

Now we just need a way to differentiate the headsets in code and set that automatically.

bschumacher, 2016-08-22:
Can you post the output from the code at: http://pastebin.com/ZtDMducP

Gabrock94, 2016-08-22:
@jspenc72 yes, it's an epoch+.
@bschumacher here you are: http://pastebin.com/mWWnYUYd

By the way, do you know if is there any project that uses bluetooth, in order to achive a 256Hz dump? 

Thanks guys for your help :)

bschumacher, 2016-08-22:
Awesome, I'm not aware of a way to do that at the moment.

bschumacher, 2016-08-22:
Did you pay for the "research edition"?

Gabrock94, 2016-08-22:
@bschumacher do you mean the raw EEG data version?

bschumacher, 2016-08-22:
yes

Gabrock94, 2016-08-23:
No, that one is a "non-EEG" one

closed 2016-10-04 by bschumacher