openyou/emokit #168: Split C and Python Libraries into different repos?

qdot, 2016-06-20:
It looks like the python library in this repo sees far more updates and attention than the C library. Should we split the C code into its own repo? That may relieve some of the bug/review pressure here, as well as letting us version the python code apart from the C. I usually just do one repo per language with my projects these days anyways.

bschumacher, 2016-06-20:
Sounds good to me.

qdot, 2016-06-20:
'k. I'll take care of the cleanup and split, will put in a pull request once it's done.

warrenarea, 2016-06-20:
i think splitting it is a good idea.

closed 2016-06-22 by bschumacher

warrenarea, 2017-11-09:
qdot (kyle) 
sorry to bother you... but I noticed Bill Schumacher has deleted his account.
any ideas whats going on there... or who has taken the reigns for emokit?

qdot, 2017-11-09:
@warrenarea Thanks for getting in touch.

Uh. Wow. That's odd. Sure enough, account seems dead, and no commits since April. Moving discussion to #279.

PR #171 (2016-06-22): Remove C Libraries
Remove all files relating to the C library, and update the README. Fixes #168.