openyou/emokit #169: EPOC+ Support

bschumacher, 2016-06-20:
Figure out gyro z, mag x, y, z, accel x, y, z mapping, make sure other sensor mappings are still valid as well.

jspenc72, 2016-06-24:
Hey anything I can do to help with Epoc + support?
I don't have my headset yet so I can't test it but wondered where how close you are to supporting the Epoc +? Have you had a chance, or do you need access to a subscription to the official Pure EEG Software so you can compare?

jspenc72, 2016-06-24:
Does this branch really support the Epoc +? Has anyone had a chance to try it?

https://github.com/a455bcd9/emokit

Noticed it within another issue?

bschumacher, 2016-06-24:
@jspenc72 Just started working on it, being able to compare would definitely help. If you haven't bought it yet though I would save your money.

I doubt it, doesn't have the 9axis gyro/accel/mag sensor in code yet, they gyro offset for x, y and the UD in the serial number are things that appear different to me at a glance.

jspenc72, 2016-06-24:
What would you recommend instead?
Open BCI?

mdtdev, 2016-06-24:
I've got some EPOC+'s around if I can help in some way with this, like maybe some testing, but I can only access them on certain days at work as they are being actively used for experiments. For the experiments we're using our legacy/grandfathered copy of the Emotiv Testbench which we get to keep using because we paid Emotiv a few grand a year or two ago. :disappointed: 

@jspenc72 Emotiv is not exactly a research friendly company for a variety of reasons. Open BCI is the obvious next choice--we just put in an order for a few development kits and a couple of headset parts kits, but we don't have the experience with them yet. The other options depend a lot on your price point. Nothing is a cheap as Emotiv, but Open BCI is the next cheapest of which I am aware.

jspenc72, 2016-06-24:
I have already bought the headset so Ill be in touch when it arrives...
I require ability to collect repeat data sets using multiple devices in order to accomplish my goals. So I am happy to buy the subscription but require the raw data to be extracted directly as you have done with this library to accomplish efficient overlaying of event tags without having to post process the data....

Which is a sure shortcoming of a research device... 

I wish they would just release the data...

jspenc72, 2016-06-24:
@mdtdev I am actually printing the Mark III right now as we speak...

jspenc72, 2016-06-24:
But as mentioned I am not so much concerned with price but collecting data from a variety of devices...

jspenc72, 2016-06-24:
@mdtdev, can you tell me the version of your test bench software?

mdtdev, 2016-06-25:
@jspenc72 Not sure of the version, but it should be one of the last builds from 2015 before they stopped supporting it. Actually I found the version, it is v.3.1.20 (Xavier). It can collect EEG and also marker signals sent through a virtual serial port.

jspenc72, 2016-06-25:
@bschumacher If it will help will be happy to set it up so you have what you need to compare data. You can pm me on Google hangouts @jspenc72 to get the details.

a455bcd9, 2016-07-18:
Hi @jspenc72, I did the fork for the EPOC+ and it was working. (I don't have the device anymore). However, I felt like the EPOC+ I was like the EPOC, except a few things, and I wasn't able to access the accelerometers and the magnetometers (if I remember well), maybe because they were not present... Hope it can help...

jspenc72, 2016-07-18:
Thanks for getting back to me on this. I finally received my epoc + so ill give it a try.

If it is working, then is this issue thread solely for adding "Gyro, Accel Support for Epoc +"?

Best,
Jesse

jspenc72, 2016-07-18:
Can anyone confirm that the Epoc + branch created by @a455bcd9 the Epoc + branch is currently works for obtaining voltage values from the latest EPOC + headset but it does provide "Gyro or Accelerometer" information?

mdtdev, 2016-07-18:
Sorry I cannot confirm it now, but I will try to give it a go as soon as I can get back to the lab and get some time not scheduled with subjects. I'll try to test it tomorrow afternoon (UTC-05:00).

jspenc72, 2016-07-18:
Thank you! @mdtdev let me know what you find.

mdtdev, 2016-07-19:
@jspenc72 I gave it an initial go:

Ok dumb question time--how do I capture the dumped signals? I can run emotiv.py as __main__ and get a dump to the screen from the EPOC+, but I don't know what to do to capture the signals (short of writing a new program in python which I cannot do just now). Is there a simple way to capture the signal so I can load it in analysis software and check the values to see if they are reasonable?

I can confirm that when I start and stop the headset data is acquired and starts and stops. Also the numbers flying by seem not unreasonable given my previous work with Emotiv's gear. But plots would help me confirm that.

Thanks!

qdot, 2016-07-19:
What platform are you on? For linux or Windows, you can use Wireshark, or Windows also has Device Monitoring Studio which is handy and comes with a 30 day trial. For OS X, well, um. I haven't captured anything on OS X in years so I'm not sure what the current up-to-date method is there.

mdtdev, 2016-07-19:
@qdot Hi! I am unfortunately on Windows and at work stuck with limited permissions from my IT department so any installs are out of the question. I'm just looking for a quick way to capture the output from testing the Emokit fork for @jspenc72 to see if it looks reasonable as proper EEG data. I believe the fork is working to pull data from the EPOC+, but want to look at the numbers to be sure. I was also hoping to do this test without actually having to write a program to capture the data.

If I can get to a computer where I can install the trial of Device Monitoring Studio, what will the device look like to it; what is the name? Will it be obvious? I don't usually work at this low of a level with hardware.

qdot, 2016-07-19:
Oh, yeah, it'll have a name and what not, but honestly if we can just get you a dump with the python scripts, that might be best. If I have time I'll take a look at the fork and see if I can help.

AlessioLanini, 2016-07-30:
Hi I bought an  EMOTIV EPOC+ and i ran   emotiv.py  ( https://github.com/a455bcd9/emokit )script on winwows 7
EPOC+ firmare version is shown as 0x30XX and dongle 0x6ff , I didn't update with last firmware yet.

[image]

I dont understannd if output is correct or not.
I think there s something wrong. 

Packets Received: 5866 Packets Processed: 3687
Y Reading: -94 Quality: 0
F3 Reading: -1207 Quality: 17
F4 Reading: 6766 Quality: 6
P7 Reading: 7297 Quality: 0
FC6 Reading: 7270 Quality: 5
F7 Reading: 2376 Quality: 23
F8 Reading: -7345 Quality: 20
T7 Reading: -6570 Quality: 9
P8 Reading: 3757 Quality: 10
FC5 Reading: -2873 Quality: 27
AF4 Reading: 7142 Quality: 15
Unknown Reading: 228 Quality: 28
T8 Reading: -1858 Quality: 25
X Reading: 55 Quality: 0
O2 Reading: -3886 Quality: 13
O1 Reading: 9 Quality: 8
AF3 Reading: -3506 Quality: 15
Battery: 2
Packets Received: 5869 Packets Processed: 3689
Y Reading: -39 Quality: 0
F3 Reading: 7189 Quality: 17
F4 Reading: -4216 Quality: 6
P7 Reading: -2903 Quality: 0
FC6 Reading: -7202 Quality: 5
F7 Reading: -8092 Quality: 23
F8 Reading: 2917 Quality: 20
T7 Reading: 8137 Quality: 1
P8 Reading: 1533 Quality: 10
FC5 Reading: 1507 Quality: 27
AF4 Reading: 1275 Quality: 15
Unknown Reading: 96 Quality: 20
T8 Reading: -3838 Quality: 25
X Reading: 39 Quality: 0
O2 Reading: -4759 Quality: 13
O1 Reading: 6837 Quality: 8
AF3 Reading: -6546 Quality: 15
Battery: 2
Packets Received: 5872 Packets Processed: 3692
Y Reading: -39 Quality: 0
F3 Reading: 2226 Quality: 17
F4 Reading: 1507 Quality: 6
P7 Reading: -3214 Quality: 0
FC6 Reading: 4461 Quality: 5
F7 Reading: -1995 Quality: 23
F8 Reading: 7426 Quality: 20
T7 Reading: -1890 Quality: 1
P8 Reading: -2100 Quality: 10
FC5 Reading: -5676 Quality: 27
AF4 Reading: 2137 Quality: 15
Unknown Reading: 62 Quality: 26
T8 Reading: 5933 Quality: 25
X Reading: -55 Quality: 0
O2 Reading: -4397 Quality: 13
O1 Reading: -5107 Quality: 8
AF3 Reading: 7287 Quality: 15
Battery: 2
Packets Received: 5875 Packets Processed: 3693
Y Reading: 42 Quality: 0
F3 Reading: -2779 Quality: 17
F4 Reading: 6331 Quality: 6
P7 Reading: 6850 Quality: 0
[... 257 lines cut ...]
O2 Reading: 1813 Quality: 13
O1 Reading: -6014 Quality: 8
AF3 Reading: 3920 Quality: 21
Battery: 2
Packets Received: 5917 Packets Processed: 3724
Y Reading: -104 Quality: 0
F3 Reading: -3404 Quality: 17
F4 Reading: -1991 Quality: 11
P7 Reading: 6314 Quality: 0
FC6 Reading: 2719 Quality: 5
F7 Reading: 6219 Quality: 3
F8 Reading: 5906 Quality: 2
T7 Reading: 6922 Quality: 14
P8 Reading: 6613 Quality: 10
FC5 Reading: -4951 Quality: 27
AF4 Reading: 7336 Quality: 15
Unknown Reading: 51 Quality: 26
T8 Reading: -5033 Quality: 25
X Reading: 11 Quality: 0
O2 Reading: 610 Quality: 13
O1 Reading: 429 Quality: 8
AF3 Reading: -1558 Quality: 21
Battery: 2

Process finished with exit code 1

ppasler, 2016-09-21:
I just checked out the current master and tried my EPOC+ under Windows.

Here my results: 

- emokit.py: works for me, but Z-reading has no value (Z Reading: ?) and battery level is always 0
- example_export: problems wit ":" in filename (#187)
- example.py: AttributeError: 'NoneType' object has no attribute 'gyro_x'
- render.py: does exit sometimes, didn't figure out when and why

Here are some recordings with alle sensors green (control panel sais so).
[emotiv_dump.csv.txt](https://github.com/openyou/emokit/files/484756/emotiv_dump.csv.txt)
Just to see if you guys have the same data area

@jspenc72: I can confirm, that the battery level is shown and gyro values are always zero in the branch from @a455bcd9.

I would be happy if there is a way to confirm the values provided by emokit. As the Premium SDK does not work at all (values around 4000) and the license model for the [pure.EEG](https://www.emotiv.com/product/emotiv-pure-eeg/) (pay per record) is way to expensive, this will be difficult task.

mattrjohnson, 2016-10-02:
@ppasler I have here what appears to be a EPOC+ (at least, it says so on the device -- I did not do the original order so I'm not 100% sure what the person who ordered it intended to get).

Regarding the issues you noted -- I think these are all intrinsic to the code and not specific to the EPOC+, except for the battery issue (which doesn't happen for me -- battery level is fine). There is nothing in the code that actually attempts to get the Z-reading.

As far as I can tell in my testing so far -- assuming that Emotiv has not actually yet implemented many of the + features of the supposed EPOC+, as others have noted -- the only major thing that might be relevant from @a455bcd9 's branch is the change in offset of the X/Y values (from 105/106 to 127). I didn't have the same issues with serial numbers and such that @a455bcd9 apparently did -- my so-called EPOC+ was recognized as such just fine from the code in this branch.

Angelacho, 2016-11-30:
Hi! First of all I would like to thank you all for your efforts in this project. I'm working for a project funded by the european union and although we have the EPOC+, it is not possible for us to access the raw data without the premium SDK or the Pure EEG software. The biggest problem is that the fund for this project doesn't allow us to buy software, so your solution has been a relief. 

I have installed emokit with all the libraries needed in windows, but when I run emokit-master\python\example.py it takes about 3 minutes to process only 6 packages. On the other hand, usually an exception occurs and the execution stops. I have attached the screenshot:

[image]

At the same time, quality values that appear doesn't seem to concord with what I can see on the Emotiv Xavier ControlPanel (all of them appear as 0 while some of them were green). 

Thank you very much again.

Ángel.

ppasler, 2016-11-30:
Hey @Angelacho,
I've also seen this behaviour, but not with the latest revision. How did you install emokit?
Try cloning the project and install the current `master` branch. This should solve the issue and provides other quality values than 0 (but still not the right ones, see #205).

mattrjohnson, 2016-11-30:
@Angelacho This is not really an effective fix for your problems -- nor an advertisement for our solution -- but if you want to try it as a debugging step, you could try our re-written 'emofox' project that I mentioned a bit in the issue #164 thread.

It uses the same basic code as emokit (as of late September / early October, anyway... looks like there has been some activity since then and I haven't looked yet to see how much has changed), but it is heavily stripped down to a bare-bones implementation. As such it is much less generally useful than emokit (we really wrote it just for ourselves) but it runs much faster on Windows, which was one of our main motivations.

(It does do a graphical plot of the incoming EEG signal, which could slow things down if you're working on a not-so-fast system... but you could basically comment out all the graphical code and replace it with a simple text interface in just a few lines of code, if you were so inclined.)

bschumacher, 2016-11-30:
Try changing the is_research option to True

bschumacher, 2016-11-30:
@Angelacho forgot to tag you 95% sure it's the fix for your problem. Your values are way off. I would advise you of course to stick to the latest github master, although 'easy_install emokit' should be pulling the latest version.

warrenarea, 2016-12-01:
Angelacho, perhaps tell us what versions of gevent and what version of greenlet you are using.  In the past, I had issues where if i installed the newest versions of gevent, it would crash in possibly a similar manner.  Last I checked the dependancy versions were:

pywinusb 0.2.9
pycrypto 2.6
gevent 1.0.1
greenlet 0.4.2 (or Newest 0.4.5)

warrenarea, 2016-12-01:
https://github.com/CymatiCorp/CyKit 
you might check there for step by step installation instructions, and links for the files.

warrenarea, 2016-12-01:
also, your error indicates that its trying to access native threading... 
so its possible that maybe it went to try to use gevent threading, and reached
for the native threading instead.  which might indicate that gevent is not
installed correctly. 

just a guess.

Angelacho, 2016-12-01:
Thank you very much for your answers. Yes, I have the current master branch installed, although I tried to pull the latest version again just in case and the problem still remains. I have also changed the is_research option to True in the initial values in emotiv.py and I don't see any difference. 

These are the versions of the libraries I have installed right now in Python 2.7:

emokit 0.0.7 (?? hope is the latest one)
pywinusb 0.4.1
future 0.16.0
pycrypto 2.6.1
gevent 1.1.2
greenlet 0.4.10

The Epoc + I'm using was acquired around June. 
I will try emofox just to see any difference.

One last thing. Yes, I'm in a slow machine and, although the write param is False, after each time I run example.py a csv is written. Even if it worked, should I expect the slow performance I commented before?

Thanks!

Angelacho, 2016-12-01:
I just tried emofox and seems that it works! I still need to test the reliability of the data but it is a good step. Thanks!

Angelacho, 2016-12-01:
Hey! I just tried to test the data in emofox. The program seems to work, but the data that it is acquiring doesn't seem to be reliable. When I use only two sensors (AF3, AF4) and acquire the signal correctly acording to the Control Panel, I'm not able to see any difference with the rest of sensors after plotting the signal (no eye blinks, no difference in values [all of them range between 8000 and -8000]). I have also tried changing the value of g_is_research to True but still having the same results.

bschumacher, 2016-12-01:
Definitely should not be writing a CSV. I'll have a look when I get a chance to see if I can reproduce your issues.

mattrjohnson, 2016-12-01:
@Angelacho Well, glad to hear it sort of worked a little. At the very least, maybe that helps you narrow down your issues with the main emokit package. emofox still relies on pygame (for graphics) as well as pywinusb and pycrypto, but it eliminates the other dependencies. So I would be inclined to agree with others that gevent/greenlet might be the most likely culprits for your emokit woes.

As for emofox itself... I'm not sure exactly what you're saying the problem is? (I read your most recent comment, I'm just not quite sure I understand.)

Anyway, it is quite possible that it would take some tweaking to get emofox working on other setups... as I say in the emofox notes, it is a very early product and really not designed for general use (at least, not yet). Our focus was first on getting things running fast and reliably in our lab, and that's as far as we've gotten. Although I would expect that if it failed on other people's setups, it would be most likely to fail in the initial setup stage (e.g. recognizing that a headset was attached)... that is the hardest part of the process.

If you want help debugging emofox, feel free to attach some sample data or further info and I can try to figure out what the problem is. Although you may find it more productive in the long run to work on debugging the more generally useful emokit instead. Still, if you want to use emofox, you are more than welcome... and I believe I have it set up on Bitbucket to handle pull requests, so if you make any useful edits to emofox in the debugging process that might be worth passing along to others, please feel free to share!

bschumacher, 2016-12-01:
It is unlikely that gevent/greenlets are the cause as they are no longer used in emokit.

mattrjohnson, 2016-12-02:
@bschumacher Ah... sorry about that. Looks like that change must have happened right after I started work on my splinter project and stopped paying attention to emokit updates. And since others had mentioned gevent/greenlet as possible culprits (which agreed with my own experience that those modules tended to be among the more trouble-causing elements of the old emokit), I assumed that info was still current.

Angelacho, 2016-12-02:
@mattrjohnson I think that it is not a matter of emofox as the code runs perfectly on my machine. The problem is with the information I'm acquiring from each sensor. It seems like if the decryption process is not working properly. It should be interesting to see if the new epoc+ devices have a different key.

bschumacher, 2016-12-03:
@Angelacho That is correct, the values are not being decrypted properly for some reason.

bschumacher, 2016-12-03:
If you could run and give the output of: python hid_info.py

Angelacho, 2016-12-05:
@bschumacher Here you have:

[image]

bschumacher, 2016-12-05:
If you look in device manager is the serial number the same or has it been mangled?

Angelacho, 2016-12-05:
@bschumacher yes, it is  UD2016010300186D

warrenarea, 2016-12-05:
UD most likely stands for "Universal Dongle" 
and 2016 01 03, being the date of purchase.

Rookie64v, 2017-01-13:
Any news on this issue? I have a very similar problem with encryption not working properly (killing the battery code makes the program work but output pseudo-random data). Here's my hid_info output:
`-------------------------
Mouse, deleted it as probably not relevant
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160303001ECC
usage, 0
path, 0003:0007:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160303001ECC
usage, 0
path, 0003:0007:01
product_string, Brain Computer Interface USB Receiver/Dongle
************************************************************
! Please include this information if you open a new issue. !
************************************************************
`
System is kUbuntu 16.04 64 bits, processor is i7 (3610QM ?) if that matters.
`uname -a:
Linux gnam 4.4.0-53-generic #74-Ubuntu SMP Fri Dec 2 15:59:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

I tried putting the serial number in and out while calling the Emotiv() class, fiddling with is_research parameter and commenting-uncommenting in and out the different options of the epoc.rules file (in all 8 combinations, assuming the command supposed to update udev rules worked), but none of them yielded results.

xiangzhang1015, 2017-01-17:
@bschumacher Hi, I had collected some data by emotiv equipment, as the result, I got the 33 numbers per sample row as output.

I have two problems, 
1. the _announcement_ (https://github.com/openyou/emokit/blob/master/doc/Announcement.md)said it only has 256 bits, which is 32 bytes(the 15th always zero), but I got **33 bytes** with the first one is always zero, why?

2. I thought maybe it's the fault of transfer from binary to decimal, so I ignore the first number(always zero ), and **transfer the last 32 numbers(decimal) to binary and then transfer them back to decimal again** follow the queue from the _Emotiv EEG Protocol Documentation_(https://github.com/openyou/emokit/blob/master/doc/emotiv_protocol.asciidoc). But it still looks not right.( I am using deep learning to classify the different eeg signal in open and close eye, my code performs great when I use one public EEG database, but performs terrible when I use the data collected by emotiv, this is why I believe the data is still not right.)
[image]

3. the announcement said the sampling rate is 128Hz, but I only got **40 Hz**. why?

I am a student focus on the classification of EEG data, and these question means a lot to me. Please help me. You can email me directly to 1015xzhang@gmail.com. Thank you very much.

xiangzhang1015, 2017-01-17:
@bschumacher To make the question clear, I want to know is the bit index in the picture is correct? Should I transfer the 32 numbers to binary again and transfer them back to decimal follow the bit index?? Thanks very much.

BTW, there are 3 question in the last comment.

xiangzhang1015, 2017-01-17:
@ppasler  Hi, I got the same EEG data as you, the first list is 0, 32 another lists after it. But there are only 14 EEg collection nodes, which 14 lists number are the EEG data?  The index_bit only has 20 items (https://github.com/a455bcd9/emokit/blob/master/doc/emotiv_protocol.asciidoc#packet-layout), I can coorespond them with 32 output lists numbers.
Please tell me(1015xzhang@gmail.com) how you distinguish them, thank you.

bschumacher, 2017-01-18:
1) I believe so but checking the source is the best way.

2) The zero gets stripped, sometimes it's there other times not. Not sure on it's meaning.

3) Without more details it's hard to say, you should be getting 128 reports per second from the headset. If you're using some other metric that could be why. If you implemented your own version it's possible you're decrypting in your reading thread which would slow it down.

Here is a more detailed get_level function, as I understand it could be wrong:

    def detailed_get_level(data, bits):
        """
        Returns sensor level value from data using sensor bit mask in micro volts (uV).
        """
        level = 0
        print("Begin Sensor Level Calculation")
        print("Data: {}".format(data))
        print("Bits: {}".format(bits))
        for i in range(13, -1, -1):
            print("Iteration: {}".format(i))
            print("Level: {}".format(level))
            level <<= 1
            print("Shift left level by one bit: {}".format(level))
            bit_at_index = bits[i]
            print("Bit Index: {}".format(bit_at_index))
            b = (bit_at_index // 8) + 1
            print("Floored(ABS or Integer) Quotient of bit and 8 plus one: {}".format(b))
            o = bit_at_index % 8
            print("Remainder of bit and 8: {}".format(o))
            if sys.version_info >= (3, 0):
                data_at_bit_index_divided_by_8_plus_one = data[b]
                print("Data at bit index divided by 8, plus one: {}".format(data_at_bit_index_divided_by_8_plus_one))
                data_shifted_right_by_remainder = data_at_bit_index_divided_by_8_plus_one >> o
                print("Data shifted right by remainder: {}".format(data_shifted_right_by_remainder))
                bitwise_of_data_shifted_right_and_one = data_shifted_right_by_remainder & 1
                print("Bitwise of data shifted right and one: {}".format(bitwise_of_data_shifted_right_and_one))
                level |= bitwise_of_data_shifted_right_and_one
                print("New level value with bitwise value added to level: {}".format(level))
            else:
                data_at_bit_index_divided_by_8_plus_one = ord(data[b])
                print("Data at bit index divided by 8, plus one: {}".format(data_at_bit_index_divided_by_8_plus_one))
                data_shifted_right_by_remainder = data_at_bit_index_divided_by_8_plus_one >> o
                print("Data shifted right by remainder: {}".format(data_shifted_right_by_remainder))
                bitwise_of_data_shifted_right_and_one = data_shifted_right_by_remainder & 1
                print("Bitwise of data shifted right and one: {}".format(bitwise_of_data_shifted_right_and_one))
                level |= bitwise_of_data_shifted_right_and_one
                print("New level value with bitwise value added to level: {}".format(level))
        return level

JulyMorning, 2017-10-13:
hello there, gentlemen. I'm considering to buy EPOC+, but, you know, SDK prices are unbelievable. Does the current version of the library supports it? It means a lot for me, so I hope you recommend me something.

warrenarea, 2017-10-13:
I can say that the Developer SDK version, does support the reading of the raw data,
but the authorization setup is a little hokey. (For python anyways)

Because it claims that you have 30 minutes of uninterrupted data, but you have to
keep authorizing it, basically every time you want to run your program... maybe that
is the way its supposed to be.

warrenarea, 2017-11-07:
https://github.com/CymatiCorp/CyKITv2