openyou/emokit #175: Rename emokit to emokit-py?

qdot, 2016-06-22:
So, now that emokit is all python, should the repo name reflect this? I'll note that this won't break links/forks/etc, github is surprisingly good about making sure links redirect, both for web and git repo updates (just had to do this on some other repos of mine).

bschumacher, 2016-06-22:
I don't really have a preference on this.

mdtdev, 2016-06-22:
Total outsider here just observing, so fwiw, but I think sticking with the original name is better. It is certainly what I was told to search for originally, and as long as you provide a link to emokit-c, I think that would be best.

qdot, 2016-06-22:
Yeah, @daeken's original emokit code was python, I was the one that added the C to it. Since we've got updates in the README about where to get the C library, I'll just hope we're ok with that. We'll just leave it be for now.

closed 2016-06-22 by qdot

warrenarea, 2016-06-23:
also, since this repository is starting out 'fresh', do you think it should be upgraded to Python 3
as well? There is no real advantage to upgrading it, and it is something of a headache with the
stricter handling of UTF strings. The only benefit though now would be, is that you could use 
the updated version of gevent, where perhaps new gevent features could be utilized. 

[http://www.gevent.org/whatsnew_1_1.html](url)
On that note, it does appear they have fixed the gevent.kill, which i seem to remember it 
used to have issues with killing the greenlet processes sometimes.

qdot, 2016-06-23:
@warrenarea Just filed #176 for this, since this bug is closed.