openyou/emokit #176: Python 3 Support?

qdot, 2016-06-23:
Emokit is currently python 2. Would it be worth considering python 3 support?

qdot, 2016-06-23:
(I have no opinion on this matter, just making sure the question didn't get lost in #175)

bschumacher, 2016-06-23:
Yeah I had planned on it anyways.

jonrei, 2016-08-01:
I have a working port in python3, but I have also removed the old Crypto and replaced it with cryptography, and removed gevents in favor of just having a generator which makes it kind of a large change. Most likely I have missed some use cases/functionality, I have only tested it on my current setup (mac with epoc+).
I think the changes are too large to easily accomodate as a pull request, and py2 support would be lost too. The code is available [here](https://github.com/butternutdog/emoren/blob/master/emoren.py) if anyone is interested in somehow integrating my changes or do anything useful with it.

mdtdev, 2016-08-01:
@jonrei ...And you have confirmed that it is working with EPOC+ -- recent posts here are suggesting some potential problems with that; and I have been unable to confirm the data stream is a proper signal on mine. You are satisfied you are pulling good data? (If you are, you are awesome and saving my life!) :smile:

jonrei, 2016-08-02:
@mdtdev Yes it is EEG data from an EPOC+ bought 3 months ago, if you want to try it out there are some rudimentary instructions [here](https://github.com/butternutdog/emoren).

warrenarea, 2016-08-03:
good work. is it functional? or are you having problems with it?

qdot, 2016-08-03:
Hmm. I think I'd be ok with py2.7 compatibility, which I think this would still hit? Not like you're using async/await, and cryptography is 2.6+. Most libraries I use these days only support python 2.7+ and 3+ anyways.

jonrei, 2016-08-04:
Made a small fix to make it compatible with 2.7+, seems fully functional.

qdot, 2016-08-04:
Yeah, really need to get CI up and running to make sure we stay working across versions, but no reason to block on that right now, we've been without it for years as is heh. Wanna go ahead and PR it?

jonrei, 2016-08-05:
Created a [PR](https://github.com/openyou/emokit/pull/180)
I guess the examples should be rewritten too. As is, I think it causes a bit too much breakage.

qdot, 2016-08-06:
Well, the main reason I wanted at PR is to remind me and others that the code is there at least. Otherwise it kinda gets lost in bug comments.

PR #188 (2016-10-04): Windows python 3 fix.
Resolves #176
closed 2016-10-04 by bschumacher