openyou/emokit #183: Data written to CSV

jemSquashness, 2016-09-06:
What data is written to the CSV and in what order? 
Is the bit stream converted to decimal and plugged into the CSV file, if so, do we need to convert to binary and concatenate different bytes? 
I've compared data from the Emokit library and the actual SDK research edition and there only seems to be minor correlations without editing the data from Emokit. 
Please can anyone who's managed to use this for actual EEG readings give me a hand. 
Thanks!

bschumacher, 2016-09-06:
It is the data that constructs each individual packet in the order received, separated by commas.

It can be written before decryption or after, if the latter a serial number is required when using the reader to decrypt the data.

jemSquashness, 2016-09-06:
Is this data processed at all? Has it had the dc offset removed?

bschumacher, 2016-09-06:
The data is unprocessed.

jemSquashness, 2016-09-06:
Is it in decimal? Or some other number system?

bschumacher, 2016-09-06:
I would recommend looking at the source for the EmotivPacket class to see how the actual values are obtained and correlated to specific sensor levels.

You can't use the data without some processing.

jemSquashness, 2016-09-06:
I do understand that, the DSP isn't the problem. I'm just confused as to why the Emokit values don't correlate to the actual Sdk research values. It should be a similar process to write them to a csv file as the csv is generated from the .edf which is just a print out of packet data, right? By no correlation i mean that they're not even in similar number ranges.

bschumacher, 2016-09-06:
I've never seen the research data output, I assume it is a file that has sensor values separated by commas for each packet separated by commas. 

This is the raw data stream from the dongle and was meant more for debugging and further development of the library with new headsets than to be used as a representation of actual signal values without being processed as an emokit packet.

jemSquashness, 2016-09-06:
Ok great, thank you for your time. 
Will try and work with the decrypted stream directly and see if i can get actual sensor values. Will post any progress.

bschumacher, 2016-09-06:
That all being said I understand what you are trying to accomplish and have added it to my todo list.

bschumacher, 2016-09-06:
I updated my fork with this functionality.

It may or may not work correctly on Windows. Linux or OS X should be fine though. 

Still struggling to get my Windows VM to recognize my dongle.

jemSquashness, 2016-09-07:
Ok great, thanks again. 
Will test it out tomorrow.

commit ba1453a6 (2016-10-03): Changed run logic.

Added write_values option to constructor, which resolves #183. By writing the sensor values and quality to CSV.

Updated example with values export as it makes more sense for end users.

Removed unused time import.

Added hid_info.py to help extract data a little easier for connected hid devices, decoupled from the rest of the library.

Support on Windows may be broken, as my VM no longer allows attaching the dongle for some reason.
closed 2016-10-03 by bschumacher