openyou/emokit #185: Packets Received: 0 Packets Processed: 0

henzer-garcia, 2016-09-16:
Hi, 

I am trying to use emokit to read raw data from EMOTIV EPOC, but when I run the example_export.py any packets are received or processed. And not appear any error.
If I see the software of EMOTIV EPOC, (Emotiv Xavier Control Panel), the reception of signals is good.

This is the result:

Packets Received: 0 Packets Processed: 0
Y Reading: 0 Quality: 0
F3 Reading: 0 Quality: 0
F4 Reading: 0 Quality: 0
P7 Reading: 0 Quality: 0
FC6 Reading: 0 Quality: 0
F7 Reading: 0 Quality: 0
F8 Reading: 0 Quality: 0
T7 Reading: 0 Quality: 0
P8 Reading: 0 Quality: 0
FC5 Reading: 0 Quality: 0
AF4 Reading: 0 Quality: 0
Unknown Reading: 0 Quality: 0
T8 Reading: 0 Quality: 0
X Reading: 0 Quality: 0
Z Reading: ? Quality: 0
O2 Reading: 0 Quality: 0
O1 Reading: 0 Quality: 0
AF3 Reading: 0 Quality: 0
Battery: 0

bschumacher, 2016-10-03:
Make sure the green light is blinking rapidly on the dongle. You might try closing the control panel if you have it open at the same time.

DJuboor, 2016-10-06:
Same Problem, ^ did not solve.

bschumacher, 2016-10-06:
@DJuboor Sometimes you have to flip the switch on the headset on and off a few times I've found.

Soon an exception will be thrown instead of no activity.

bschumacher, 2016-10-08:
The master branch has a fix for this in it now, if no data is received emokit will reinit the reader until it gets data.

closed 2016-10-08 by bschumacher

DJuboor, 2016-10-10:
Seems I was also using old sourcecode from another project and was calling the data wrong. Thank you for the update though! All is working well.

bschumacher, 2016-10-10:
Good to hear.

PR #195 (2016-10-08): Fix timeout by continually restarting reader hid.
Resolves #185 #189