openyou/emokit #191: print_hid_enumerate leads to UnicodeEncodeError

ppasler, 2016-10-07:
Hi there,

the ´print_hid_enumerate´ methods leads to an UnicodeEncodeError when running on Windows 7 in german language.

```python
UnicodeEncodeError: 'ascii' codec can't encode character u'\u0409' in position 0: ordinal not in range(128)
```

The product_name should be: 
´product_name, @input.inf,%hid_device%;HID-konformes Gerät´

The Umlaut "ä" seems to be the problem.

This might be the cause: https://www.reddit.com/r/learnpython/comments/2ytaw2/how_to_deal_with_unicode_exceptions/
but using ´encode("utf-8")´ leads to a wrong character ´GerÃ¤t´ 

This is critical, as the exception breaks the ´hid_info´ module for me. Exception handling would do the job in this case.

```python
try:
    print("%s, %s" % (key, str(value)))
except UnicodeEncodeError:
    print ("%s, %s" % (key, value.encode("utf-8")))
```

bschumacher, 2016-10-07:
Are you using python2 or 3? You might try .encode("latin-1") or even .encode("iso-8859-1")

ppasler, 2016-10-08:
Hi @bschumacher,

I am working with `Python 2.7.10`.  Using `latin-1` or `iso-8859-1` leads to the same error as described above.

```python
print("%s, %s" % (key, value.encode("iso-8859-1")))
UnicodeEncodeError: 'latin-1' codec can't encode character u'\u0409' in position 0: ordinal not in range(256)
```

bschumacher, 2016-10-08:
I found this snippet in the pallets issues:
`
    import sys
    import locale

    # get console's codepage
    cp_console = sys.stdout.encoding # results in e.g. 'cp850'

    # get preferred codepage
    cp_preferred = locale.getpreferredencoding() # results in e.g. 'cp1252'

    # transform input string
    print("stdout encoding: {}".format(cp_console))
    print("locale encoding: {}".format(cp_preferred))
    
    #console_input = console_input.encode(cp_console).decode(cp_preferred)
`

What are the two encodings when you run it on your system?

ppasler, 2016-10-10:
running the snippet with IDLE it results in:
```python
stdout encoding: cp1252
locale encoding: cp1252
```
running in Eclipse
```python
stdout encoding: UTF-8
locale encoding: cp1252
```
By the way, both environments result in the same error.

I do not want to exclude that the error is something that only happens with my explicit encoding settings (Windows sucks in this way, I know).

bschumacher, 2016-10-10:
Lets try adding this import: 

    import locale

and do this instead:
    
    print("%s, %s" % (key, str(value).encode(locale.getpreferredencoding())))

bschumacher, 2016-10-10:
If you could checkout that branch and let me know if that fixes it for you.

ppasler, 2016-10-11:
As the error happen when calling `str` the `encode` does not solve it.

Using `unicode` instead of `str` solves the issue for me:
```python
print("%s, %s" % (key, unicode(value)))
```
Sorry for keeping you up from "real" problems...

I would push the solution myself to the branch, but I have no permission :)

bschumacher, 2016-10-11:
I believe that would break py3 though.

bschumacher, 2016-10-11:
Ok I updated that branch again, if you would please give it a try.

ppasler, 2016-10-11:
Indeed it would break Python 3 - good point.
Sorry, did not work, as the input type which causes the error is `unicode` and not `str`.

bschumacher, 2016-10-11:
Was it the same error message cannot encode character with cp1252?

ppasler, 2016-10-11:
Yes, because the thing causing the error is not ``type: str` it's `type: unicode` so the if statement does not work.

bschumacher, 2016-10-11:
I understand now, I think we got it this time.

ppasler, 2016-10-11:
Cool stuff!

* As you add the `unicode` function for python 3, you could use it and remove the outer if statement.
* `unicode` works with all types, so no need to have the inner if statement, I think.

In my opinion, this should be everything inside the loop
```python
            if not sys.version_info >= (3, 0):
                print("%s, %s" % (key, unicode(value)))
``` 
I am not able to test with python 3, but for me it works fine this way!

bschumacher, 2016-10-11:
Yeah that makes more sense, I've updated that branch again. Could you please see if it still works?

ppasler, 2016-10-11:
Works like a charm! Thanks for fixing.

PR #196 (2016-10-10): Locale fix.
Resolves #191, maybe.
closed 2016-10-11 by ppasler