openyou/emokit #192: Sampling rate too low

pranaydeep-af, 2016-10-07:
Using the device with the EPOC Headset, All packets are processed and none are lost, the device sampling rate is still too low. Close to around 50-55Hz. Could there be a configuration mistake? Or is the device faulty?

bschumacher, 2016-10-07:
Are you running the example or something else?

pranaydeep-af, 2016-10-08:
I have tried running the various codes, one of them was the example. No luck with any of them. I am on Ubuntu Gnome 16.04 and using the hidraw bindings.
Thanks for the help

bschumacher, 2016-10-08:
If you could use the code that is mentioned above and tell me what it says the sampling rate is.

I can confirm on my system that it should be 128.

bschumacher, 2016-10-19:
Did you happen to see if this gave you a value other than 127-128 after running for a few seconds?

bschumacher, 2016-10-23:
If you're still seeing abnormal performance please open another issue.

closed 2016-10-23 by bschumacher