openyou/emokit #201: Device not found OS X 10.11.6

pabloriera, 2016-10-23:
This is the output of the test code. Any steps to follow to check if the problem is not with emokit?  Thanks

```
Emotiv Systems Pty Ltd
Receiver Dongle L01
b'USB_1234_ed02_4100000'
4660
60674
sN20131213002732
```

```
Please include this information if you open a new issue.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Users/Shared/anaconda/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/Users/Shared/anaconda/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/Shared/anaconda/lib/python3.5/site-packages/emokit-0.0.6-py3.5.egg/emokit/emotiv.py", line 191, in run
    self.initialize_reader()
  File "/Users/Shared/anaconda/lib/python3.5/site-packages/emokit-0.0.6-py3.5.egg/emokit/emotiv.py", line 119, in initialize_reader
    self.reader = EmotivReader()
  File "/Users/Shared/anaconda/lib/python3.5/site-packages/emokit-0.0.6-py3.5.egg/emokit/reader.py", line 59, in __init__
    self.setup_platform[self.platform]()
  File "/Users/Shared/anaconda/lib/python3.5/site-packages/emokit-0.0.6-py3.5.egg/emokit/reader.py", line 196, in setup_not_windows
    raise Exception("Device not found")
Exception: Device not found
```

bschumacher, 2016-10-23:
Try this repo: https://github.com/CerebralPower/emokit

pabloriera, 2016-10-23:
Thanks,

It didn't work with that repo either, but I notice this message (it also was previously, but I missed)

`Emotiv IsEmotivError  <class 'AttributeError'> 'hid_device_info' object has no attribute 'vendor_name' <traceback object at 0x101aa8b88>  :  'hid_device_info' object has no attribute 'vendor_name'`

bschumacher, 2016-10-23:
Try that latest commit please.

pabloriera, 2016-10-23:
Amazing, thanks a lot.

Now it worked, but the Gyro data didn't responded to movements. I will check a little more.

PR #202 (2016-10-23): Unicode fix
Resolves #201
closed 2016-10-23 by bschumacher