openyou/emokit #204: Can emokit record at a higher sample rate?

windweller, 2016-11-23:
We are using this library to extract raw eeg data, and things have been going very well! However, we can't help to notice that the data we get are roughly 38 data points (samples) per second. Based on the website, the sample rate is much higher: 128 or 256 samples per second (https://www.emotiv.com/epoc/, under tech specification).

I'm wondering if this is a problem with the transmission, or the headset, or the library?

bschumacher, 2016-11-23:
If you run the example.py is the sample rate also low?

Which OS and Python version are you running it on?

windweller, 2016-11-25:
yes, it is run with example.py, the OS is Win 7 professional SP1, and Python version is 2.7.12

ppasler, 2016-11-28:
I quite achieve 128 data points per second with Win7 and python 2.7. @windweller have you tried a faster machine?
Just for the record: The sample rate (128 or 256) for the EPOC+ has to be configured with the Xavier Control Panel.

Morgan243, 2016-12-03:
For those debugging this - I get roughly 128 HZ, and it seems slightly more consistent if I disable the writer. I'm on Linux with a 4 core Intel i5-3320M and an SSD.

windweller, 2016-12-05:
We did some testing, and used a fork of this repo that does give us 128Hz recording, however, once we added the code to have the program write csv output to a file, the rate dropped to 40Hz.

So we are hypothesizing that this could be a file IO blocking issue!

bschumacher, 2016-12-05:
@windweller Yeah it makes sense looking back through it I thought I had fully implemented multithreading but failed to do so on the writer module.

I will get it fixed.

bschumacher, 2016-12-05:
Alright if you could pull down this repo: https://github.com/CerebralPower/emokit

Let me know what you think, made some changes.

bschumacher, 2016-12-05:
Make sure you are not using Python3, it is not stable as Python2 when exporting.

I'm gonna work on that but no idea when I'll get around to it.

windweller, 2016-12-06:
YES! The repo worked :) Thank you!

bschumacher, 2016-12-06:
No problem, I'll get it merged in here in a bit. Have to wait for headset to charge up again.

closed 2016-12-06 by bschumacher

ppasler, 2016-12-17:
I can confirm, that the latest master writes values with 128Hz (console and file both on). 
Thanks @bschumacher

PR #213 (2016-12-06): Issues
Lots of stuff updated here.

Might resolve #211 

Might resolve #209 

Might resolve #207 

Might resolve #205 

Might resolve #204