openyou/emokit #205: strange quality values

ppasler, 2016-11-28:
With the latest version ([9628c1f](https://github.com/openyou/emokit/commit/9628c1f5db98b18c40266e7d07260c8ed84a49e7)) there are strange quality values. I am using a EPOC+ and Win7.

```
Packets Received: 225 Packets Processed: 224
Y Reading: 23 Quality: 0
F3 Reading: -20 Quality: 1552
F4 Reading: 7 Quality: 7256
P7 Reading: 27 Quality: 3256
FC6 Reading: -15 Quality: 3368
F7 Reading: 0 Quality: 648
F8 Reading: -78 Quality: 2320
T7 Reading: 64 Quality: 1048
P8 Reading: -47 Quality: 560
FC5 Reading: -452 Quality: 2784
AF4 Reading: 137 Quality: 1528
Unknown Reading: 109 Quality: 7248
T8 Reading: -64 Quality: 2920
X Reading: 21 Quality: 0
Z Reading: ? Quality: 0
O2 Reading: -63 Quality: 2000
O1 Reading: -42 Quality: 1872
AF3 Reading: -108 Quality: 2288
Battery: 82
Sample Rate Rx: 52 Crypto: 52
packet {'quality': 0, 'value': 21
```

The last "working" revision was [9994035](https://github.com/openyou/emokit/commit/9994035214c085836fdef98d6d4ed1bb1d999974). After that the values all have been zero for some commits, up to this behavior.

bschumacher, 2016-12-01:
Yes, this is the actual values instead of them being divided based on the model of headset. Since they keep changing this on us it is easier to report the actual data and provide a helper function to parse it further if required.

You can use the get_quality_scale() function of an EmotivPacket object to get the approximate 0-4 or whatever it was.

ppasler, 2016-12-02:
I see, but these kind of "API changes" are not transparent to users. There should be some kind of documentation. I would suggest the call of `get_quality_scale()` should be default for now.

bschumacher, 2016-12-06:
The reason is people were saying they were getting a zero value, with the way they change these values per headset it's hard to tell if they are really getting a zero value of if we are using the wrong division scheme.

I'll see if I can come up with something better.

PR #213 (2016-12-06): Issues
Lots of stuff updated here.

Might resolve #211 

Might resolve #209 

Might resolve #207 

Might resolve #205 

Might resolve #204
closed 2016-12-06 by bschumacher