openyou/emokit #206: Add Z-value to packet

ppasler, 2016-11-28:
As the EPOC+ provides X, Y, and Z-values, it would be cool to have all three values.
At this point the Z-value shows a ``?`` as value.

ppasler, 2016-12-02:
I am not really deep into the binary value processing. Could anyone explain how the binary EPOC signal is decoded, as there are more values to be fetched: 

> The 9-axis inertial motion sensor include:
> 
> * 3-axis gyroscope
> * 3-axis accelerometer
> * 3-axis magnetometer

[EPOC knowledge base](https://emotiv.zendesk.com/hc/en-us/articles/201390999-Why-do-I-need-9-axis-inertial-motion-sensor-)

xiangzhang1015, 2017-01-17:
Hi, can I know in the 33 lists outputs, which 14 are the 14 EEG channels? Thanks very much.