openyou/emokit #218: Reader Thread Keeps Restarting

lubricationTheory, 2017-01-10:
I tried running the example.py.

The script keeps restarting the reader thread. Here is the output I get from the script:

> Initializing Reader Thread...
> Reader detected serial number: UD2016010300144E
> Reader Thread Restarted
> Reader stopping...
> Reader stopped...
> Restarting Reader
> Initializing Reader Thread...
> Reader detected serial number: UD2016010300144E
> Reader Thread Restarted
> Reader stopping...
> Reader stopped...
> Restarting Reader
> Initializing Reader Thread...
> Reader detected serial number: UD2016010300144E
> Reader Thread Restarted
> Reader stopping...
> Reader stopped...

Here is the result of running the hid_info.py script: 

> -------------------------
> hid_handle, None
> parent_instance_id, 9
> report_set, {}
> usages_storage, {}
> _HidDevice__input_report_templates, {}
> _HidDevice__reading_thread, None
> serial_number, DQ63277YT71F94RACCC
> product_name, Keyboard / Boot
> _HidDevice__input_processing_thread, None
> device_path, \\?\hid#vid_05ac&pid_0291&mi_01&col02#7&163aea46&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
> vendor_id, 1452
> _HidDevice__button_caps_storage, []
> _HidDevice__open_status, False
> _input_report_queue, None
> version_number, 369
> product_id, 657
> _HidDevice__evt_handlers, {}
> instance_id, HID\VID_05AC&PID_0291&MI_01&COL02\7&163AEA46&0&0001
> ptr_preparsed_data, None
> _HidDevice__raw_handler, None
> vendor_name, Apple Inc.
> hid_caps, None
> -------------------------
> hid_handle, None
> parent_instance_id, 10
> report_set, {}
> usages_storage, {}
> _HidDevice__input_report_templates, {}
> _HidDevice__reading_thread, None
> serial_number, DQ63277YT71F94RACCC
> product_name, Trackpad / Boot
> _HidDevice__input_processing_thread, None
> device_path, \\?\hid#vid_05ac&pid_0291&mi_02&col02#7&2dfe768b&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
> vendor_id, 1452
> _HidDevice__button_caps_storage, []
> _HidDevice__open_status, False
> _input_report_queue, None
> version_number, 369
> product_id, 657
> _HidDevice__evt_handlers, {}
> instance_id, HID\VID_05AC&PID_0291&MI_02&COL02\7&2DFE768B&0&0001
> ptr_preparsed_data, None
> _HidDevice__raw_handler, None
> vendor_name, Apple Inc.
> hid_caps, None
> -------------------------
> hid_handle, None
> parent_instance_id, 9
> report_set, {}
> usages_storage, {}
> _HidDevice__input_report_templates, {}
> _HidDevice__reading_thread, None
> serial_number, DQ63277YT71F94RACCC
> product_name, Keyboard / Boot
> _HidDevice__input_processing_thread, None
> device_path, \\?\hid#vid_05ac&pid_0291&mi_01&col03#7&163aea46&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
> vendor_id, 1452
> _HidDevice__button_caps_storage, []
> _HidDevice__open_status, False
> _input_report_queue, None
> version_number, 369
> product_id, 657
> _HidDevice__evt_handlers, {}
> instance_id, HID\VID_05AC&PID_0291&MI_01&COL03\7&163AEA46&0&0002
> ptr_preparsed_data, None
> _HidDevice__raw_handler, None
> vendor_name, Apple Inc.
> hid_caps, None
> -------------------------
> hid_handle, None
> parent_instance_id, 9
> report_set, {}
> usages_storage, {}
> _HidDevice__input_report_templates, {}
> _HidDevice__reading_thread, None
> serial_number, DQ63277YT71F94RACCC
> product_name, Keyboard / Boot
> _HidDevice__input_processing_thread, None
> device_path, \\?\hid#vid_05ac&pid_0291&mi_01&col04#7&163aea46&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
> vendor_id, 1452
> _HidDevice__button_caps_storage, []
> _HidDevice__open_status, False
> _input_report_queue, None
> version_number, 369
> product_id, 657
> _HidDevice__evt_handlers, {}
> instance_id, HID\VID_05AC&PID_0291&MI_01&COL04\7&163AEA46&0&0003
> ptr_preparsed_data, None
> _HidDevice__raw_handler, None
> vendor_name, Apple Inc.
> hid_caps, None
> -------------------------
> hid_handle, None
> parent_instance_id, 11
> report_set, {}
> usages_storage, {}
> _HidDevice__input_report_templates, {}
> _HidDevice__reading_thread, None
> serial_number, UD2016010300144E
> product_name, EEG Signals
> _HidDevice__input_processing_thread, None
> device_path, \\?\hid#vid_1234&pid_ed02&mi_01#7&27560a11&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
> vendor_id, 4660
> _HidDevice__button_caps_storage, []
> _HidDevice__open_status, False
> _input_report_queue, None
> version_number, 6
> product_id, 60674
> _HidDevice__evt_handlers, {}
> instance_id, HID\VID_1234&PID_ED02&MI_01\7&27560A11&0&0000
> ptr_preparsed_data, None
> _HidDevice__raw_handler, None
> vendor_name, Emotiv
> hid_caps, None
> -------------------------
> hid_handle, None
> parent_instance_id, 12
> report_set, {}
> usages_storage, {}
> _HidDevice__input_report_templates, {}
> _HidDevice__reading_thread, None
> serial_number, UD2016010300144E
> product_name, Brain Computer Interface USB Receiver/Dongle
> _HidDevice__input_processing_thread, None
> device_path, \\?\hid#vid_1234&pid_ed02&mi_00#7&f927dcc&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
> vendor_id, 4660
> _HidDevice__button_caps_storage, []
> _HidDevice__open_status, False
> _input_report_queue, None
> version_number, 6
> product_id, 60674
> _HidDevice__evt_handlers, {}
> instance_id, HID\VID_1234&PID_ED02&MI_00\7&F927DCC&0&0000
> ptr_preparsed_data, None
> _HidDevice__raw_handler, None
> vendor_name, Emotiv
> hid_caps, None
> ************************************************************
> ! Please include this information if you open a new issue. !
> ************************************************************
> Press any key to continue . . .

I am using the EPOC+ headset.

bschumacher, 2017-01-18:
The thread restarting means it hasn't received any data yet, make sure the light is blinking rapidly, try toggling the on/off switch a couple times.

xribene, 2017-02-24:
hello, 
I have the excact same problem in windows 10 , python 2.7.12, anaconda,
the light is blinking in the right way  and i have toggle the on/off switch multiple times. It seems to receiving data because it works with xavier control. It also works in windows 7. I am using the EPOC headset

[image]

xribene, 2017-03-10:
I restored my laptop to factory settings and the first thing I tried before installing any other program was to run example.py. It worked, and I made a system restore point. 
    The next day installed some other programs and after trying again the same problem appeared , reader kept stopping and restarting. So I restored again the laptop to the restore point I had make and it didn't work again.
@bschumacher  do you have any idea what the problem could be ?

bschumacher, 2017-04-05:
@xribene Crazy windows bug... I'm not sure tbh. When the blinking light on the dongle dims and begins blinking rapidly it should work. Of course it's entirely possible if you have the xavier controller access the USB device at the same time or opened before emokit that it is locking the resource preventing other programs from using it, the standard fix in that instance would be to reboot.

bschumacher, 2017-04-05:
@lubricationTheory Even if it did start, it wouldn't work reference issue #229.

closed 2017-04-05 by bschumacher

dlzln, 2021-04-27:
hello everyone,
I have the excact same problem in windows 10 , python 3.8.5, anaconda3,
when I run the example.py , it keeps "restarting reader    initializing reader thread....   reader detected serial number:UD20190115003DCE   reader thread restarted      reader stopping   " cycle appearing.  Is there anyone solve this problem? Thank you very much.