openyou/emokit #220: "Exception: Device not found" - Linux

advaitb, 2017-01-26:
Hi, 

I have a problem similar to the one listed in #201  This is what my terminal shows while running example.py:

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/advaitbalaji/Downloads/emokit-master/python/emokit/emotiv.py", line 229, in run
    self.initialize_reader()
  File "/home/advaitbalaji/Downloads/emokit-master/python/emokit/emotiv.py", line 133, in initialize_reader
    self.reader = EmotivReader()
  File "/home/advaitbalaji/Downloads/emokit-master/python/emokit/reader.py", line 58, in __init__
    self.setup_platform[self.platform]()
  File "/home/advaitbalaji/Downloads/emokit-master/python/emokit/reader.py", line 197, in setup_not_windows
    raise Exception("Device not found")
Exception: Device not found

I tried running the latest commit, but it still shows the same error :( any idea why and how to solve this

bschumacher, 2017-02-08:
It appears as though your dongle is not detected.

Ensure that you have all of the requirements installed and that your light is flashing on the dongle.

minifiredragon, 2017-04-03:
Hint-
Type sudo python example.py

Posting this here for future reference for me as well as I probably will forget.

bschumacher, 2017-04-05:
If you're still having a problem after using sudo, which I don't think is required, but maybe. Please open a new issue. If your serial number begins with UD2016 or similar your device is not going to work properly. Reference issue #229 in that instance.

closed 2017-04-05 by bschumacher