openyou/emokit #221: Full IMU

infusion, 2017-01-27:
Hi,

thank you so much for emokit! I wonder, the official SDK has an example for motion logging, which includes accelerometer, magnetometer and gyro data. You export X/Y only, is there a way to get all dimensions?

Thanks!

bschumacher, 2017-02-08:
I'm sure there is but someone is going to have to spend some time on it and do some science =).

If I had a newer headset I would hack on it myself, this is more of a gathering data thing to make sure we got it right.

What would be a great help is to record the raw data from the headset while performing the action for which would trigger a change in the data we are looking for.

Like moving a magnet past the sensor(s), accelerating the headset in a specific direction, etc.

Then perhaps one of these days someone can spend a day trying to figure it out with the data.

infusion, 2017-02-08:
I do exactly this atm, but have to pause the work for a couple weeks. But I will continue on this and bring it back to here if u want.

bschumacher, 2017-02-08:
That would be great.