openyou/emokit #222: Render.py problem: Invalid color argument

felipefpx, 2017-02-05:
I ran "python render.py" and I got this error:

No psyco. Expect poor performance. Not really...
Initializing Emokit...
Initializing Reader Thread...
Initializing Writer Thread(s)...
Initializing Crypto Thread...
Initializing Output Thread...
Crypto thread stopping. 
Reader stopping...
Reader stopped...
('EmotivRender ', <type 'exceptions.TypeError'>, TypeError('invalid color argument',), <traceback object at 0x10a6ee950>, ' : ', TypeError('invalid color argument',))

xribene, 2017-02-24:
I have the same exactly problem but in windows 7 and EPOC with bluetooth

tallhobbes, 2017-03-07:
I had this problem as well, and it seems the color needs to be a tuple, not a string. Deleting the str() wrapper around get_quality_scale_level_color(args) worked for me, since that function returns a tuple to begin with. Not sure why the string conversion is there, maybe pygame has changed its expectations? Anyway, hooray for squiggly lines!

xribene, 2017-03-10:
this worked . thank you @tallhobbes

bschumacher, 2017-04-05:
Apparently New implementation of Color was done in pygame 1.8.1.

PR #234 (2017-04-05): Fix color in render.py and faster lookups util.py.
 Resolves #222
commit b36d56bf (2017-04-05): Fix color in render.py and faster lookups util.py. Resolves #222
closed 2017-04-05 by bschumacher