openyou/emokit #223: windows 10, Reader stopping... error

xribene, 2017-02-24:
Hello,
I searched in previus issues but i did not find something similar except issue #222 
I have manage to make it work in windows 7 but it does not work in windows 10 
The error i get is that reader keeps stopping and restarting for example if I ran example.py I get this : 
[image]

the same error of course appears with every function in the python folder (example_export.py , mouse_control.py and render.py)

I thought it was a communication problem but it works good with xavier control and the leds in the usb are blinking in the right way

I use windows 10 , anaconda python 2.7.12 and i have try installing emokit in both ways
1)pip install emokit and with
2) python setup.py install
with  the same results as described

xribene, 2017-02-24:
how can i delete this issue ? after i made it i saw that there is a similar one . thanks

closed 2017-02-24 by xribene

olorin, 2017-02-24:
You can't delete it - closing it is fine.