openyou/emokit #225: threading error

firassafiyeddeen, 2017-02-27:
Hello all,

i'm getting this error when i'm trying to run the tests file on windows 10:

C:\Users\FIRAS\Desktop\eeg_data\emokit-master\python>python -m pytest tests/
============================= test session starts =============================
platform win32 -- Python 3.6.0, pytest-3.0.5, py-1.4.32, pluggy-0.4.0
rootdir: C:\Users\FIRAS\Desktop\eeg_data\emokit-master\python, inifile:
collected 6 items

tests\test_emokit.py ..
tests\test_packet.py ..
tests\test_util.py Reader detected serial number: UD20160303001E8B
..
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\FIRAS\Anaconda3\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\FIRAS\Anaconda3\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\FIRAS\Desktop\eeg_data\emokit-master\python\emokit\emotiv.py", line 298, in run
    new_packet = EmotivPacket(decrypted_task.data, timestamp=decrypted_task.timestamp)
  File "C:\Users\FIRAS\Desktop\eeg_data\emokit-master\python\emokit\packet.py", line 39, in __init__
    self.battery = battery_values[str(self.counter)]
KeyError: '187'

I have all dependencies installeds.

Is somemone having the same error?

Thanks

bschumacher, 2017-02-27:
The test was expecting the device to not be found, looks like you have to have you're going to need to use the is_research=True flag, for example to make the test work but fail in test_emokit.py:

def test_emotiv_no_headset():
    try:
        emo_test = Emotiv(is_research=True)
    except Exception as ex:
        assert (ex.message == "Device not found")

When using the library you're going to have to do the same thing most likely.

bschumacher, 2017-04-05:
This is most likely related to #229 please follow that issue instead. If you managed to solve this open a PR please.

closed 2017-04-05 by bschumacher