openyou/emokit #228: Status of current hardware support?

flarn2006, 2017-03-23:
What's the status of support for the current (as of March 2017) line of products, like the Insight? The [FAQ](https://github.com/openyou/emokit/blob/master/FAQ.md) says all headsets are supported, but #146 mentions that the Insight is unlikely to be supported any time soon, which makes me think maybe the Insight isn't included or something.

I think it's especially important that we get support for it now, because Emotiv has made the situation for raw EEG access even worse. Now, not only does it cost extra, but it's a subscription. And on top of that, they charge you depending on how many recordings you make. They're presenting (and charging for) it as if it's a service they're providing, except it's obvious they aren't providing anything except a temporary, partial relief from artificially-imposed limitations, something they shouldn't even need to give me in the first place. They're billing their customers for using their own hardware, taking money for a "service" that's really nothing more than an existing part of a product that the customer already paid for in full.

Shit like this makes me not want to buy a headset from them, not out of a desire not to support their business practices (though I do have that desire) but simply because I don't want to spend hundreds of dollars to get something that's going to behave as if I don't really own it, unless I know I'll be able to fix that issue. I want to really *feel* like it belongs to me and not the manufacturer, not just know that legally it does. And even though I'll most likely have no use for the raw EEG function, I know it's a function of the device, and unless I have unlimited access to it, I can never feel that way.

So are the current devices supported? And if not, what's the most up-to-date status on getting them to work?

bschumacher, 2017-03-24:
Someone has to buy the device(s) and provide data or submit PRs.

I don't plan on buying another headset, the first one's cheap plastic fell apart relatively quick. I also don't use it at all except to help with this.

The original headset is supported, the + is supported as much as the original. Some gyros and such are not yet.

As long as they didn't change too much the newer ones might work too, but who knows.

bschumacher, 2017-03-24:
You might consider rolling your own and using openeeg, I hear that's popular nowadays plus you might save even more money. I honestly am disappointed with the quality of the headset.

warrenarea, 2017-03-24:
Well, I was tempted to get the Insight, but if I do, I'd wait until it goes back on discount. 

I'm kind of conflicted on whether it would be worth it though, as it doesn't have half as 
many sensors, and the epoc+ has increased sensitivity.  The only benefit now is maybe its
structural integrity and gel-based sensors.

I think its interesting, whenever they show people with the insight, they always show some
pretty people jogging and having fun wearing the headset.  However everyone knows that even
the slightest movement of the headset would create artifacts, rendering any recording of physical
activities useless.  I suppose its not as attractive as the reality, of a computer nerd surrounded by 
soda cans in a secluded apartment trying to figure out ways to reverse engineer their products, so 
they don't have to pay an arm and a leg to utilize their product.

Basically, I think their whole business model is just designed to take advantage of schools and 
researchers who get grants and can afford their setup. 

Emotiv is really doing themselves a disservice though, by limiting the only people who can profit
from it, to Emotiv itself, it makes it difficult and unattractive for developers to make anything for their
devices.  Which means they sell even less devices, and are pushing people to use OpenBCI instead.

Its like I created an interface for Unity3D, but I can't release it, because I'd be giving raw access to 
anyone who used my interface, and according to Emotiv... each person who utilized my interface
would then have to buy a license from them.  ridiculous, yet they won't release their own Unity3D
interface.

I know what bschumacher is talking about though.... i'd definitely have some doubts if i knew i'd
be getting some of the cheap plastic that my Epoc was made out of.  Apparently, they said they
had an issue in the manufacturing process... which caused it to become brittle.  Supposedly that
is fixed now. 

At this point, I'm kind of just waiting for Emotiv to make the next move.... whether its reduced
prices of their products (which obviously doesn't cost that much to make), or whether they may
come out with a new and improved model... or a change to their ridiculous raw license scheme.

I do have some projects i'd like to try out, but i'm thinking now about just biding my time.

I'm not impressed with the soldering with the Epoc too... wires and components just fall off over time.
If I was a school or a university, i could just absorb the loss, but i'm already on my 2nd epoc now. lol

warrenarea, 2017-03-24:
I think i'd love to have one of those 3D printing devices.... and once they work out all of the bugs and
maybe come down a little in price, i probably will get one.

Would be a heck of a lot easier prototyping your own device.... and bill wouldn't have to worry about
cheap plastic breaking on him. lol  built ford tough!

bschumacher, 2017-04-05:
Current devices, apparently are not supported at all.

bschumacher, 2017-04-17:
As of now I would say the epoc+ is actually more supported than the old epoc.

Still working on gyro's and finding the correct method of calculating sensor signal level on the old epoc.

andy94, 2017-04-20:
Is it possible to get the raw data from an Emotiv Epoc+ (the serial number starts with UD20160503...). I don't have the Emotiv Pure EEG program. 

@minifiredragon I saw in another discussion that you changed the mode in which your device is running, from Headset Settings. Here is a print-screen of your reply: 
[image]
Is this from Emotiv Pure EEG program? If yes, is there another way to change all of these parameters without having Pure EEG? Also, if I buy Emotiv Pure EEG program, and I change the mode, is this change persistent?

In this moment, with the **latest** version of emokit, if I run render.py I get:
[image]
Obviously, this is not so accurate. Maybe, by default, my headset runs in mode epoc+ 256hz.
@minifiredragon I read that you have succeeded when you used the mode Epoc 128hz 14 bit (but first I need to figure out how to change the mode, see the first question).

What should I have to change in order to get the correct data? I could provide you with any other information.

Thank you in advance for your help!

minifiredragon, 2017-04-22:
You can only change the mode with Pure.EEG that I know of.  And yes, the change stays.

As far as what you are getting with render.py, you need to have the proper settings as per the mode your headset is in.

YikeStone, 2017-12-17:
Can anyone please tell me - how to proceed on adding support for Emotiv Insight?
I bought one for my college project and want raw eeg for SSVEP.
Pls reply ASAP as i have a deadline for the project.

warrenarea, 2017-12-17:
I have been working on support for the Epoc+ for some time now, so it has pushed
back the development for the Insight. 

Currently I am the lead developer for emokit, and I have been focusing all of my 
attention on finishing off the CyKITv2 github repository.

I still have some more work to do with the Epoc+ but I hope to start work with the
Insight, relatively soon. 

After I sort out the Insight (for CyKITv2), I will then swing back around to emokit and
reboot the repository. 

There is currently a discord server that you can get updates on
and see the most updated/live information on the progress of the projects.

[https://discord.gg/gTYNWc7](url)

JulyMorning, 2017-12-18:
@warrenarea ≈ really appreciate your time and your answers here. Could you tell us how well is EPOC+ supported for this moment? What features are fully unlocked? I mean the last version we can buy. How do you think, is the «Foc.us EEG DevKit» able to replace it technically for its price?

YikeStone, 2017-12-18:
@warrenarea, thanks for the prompt response. Is there any way I can help you with Insight?

warrenarea, 2017-12-19:
JulyMorning: well the raw data for Epoc+ is pretty much unlocked, with some margin for error. The values have been decoded to their corresponding floats, and I have produced renderings that almost match those of Pure.EEG identically.   Foc.us uses a completely different technology, so it doesn't seem fair to compare with Emotiv. Emotiv products have always been relatively fairly priced.  apart from their 'data as a service' plan. 

YikeStone: Well, It has been some time since I have had a chance to look at the data from the insight,
my initial observations kind of indicated that it used a similar method to the Epoc version, with the
possibility of different masking numbers... Which is partly why I incorporated the ability to change 'masks' 
dynamically through the CyKITv2 interface. 

I am pretty close to a 'commit' for CyKITv2, so i may try to slip something in for that too, providing
it is not overly complicated.

warrenarea, 2017-12-19:
What will help me best, is having everyone interested, on the discord server.
and then providing as many recordings of the data from EmotivPRO, with 
synchronized recordings of the data from CyKITv2.

warrenarea, 2017-12-24:
I uploaded a new revision of CyKITv2, might check it out.

YikeStone, 2018-01-16:
@warrenarea What's the status on Insight driver support for emokit? I really need the driver for my project, since deadline is approaching. In the meantime I will try and use CyKITv2.

warrenarea, 2018-01-16:
emokit has not been worked on in some time, and I haven't even begun to work with it.
currently your best option is to use CyKITv2.  Which currently supports streaming the data to 
TCP connections. (browser not required)

YikeStone, 2018-01-27:
I am using BLE connection instead of USB receiver. I want to know how get data from the GATT characteristics?

warrenarea, 2018-01-27:
[https://github.com/CymatiCorp/CyKITv2/blob/master/Documentation/Bluetooth_Development-Epoc.pdf](url)