openyou/emokit #229: Epoc+ How to identify if decryption is working?

minifiredragon, 2017-04-03:
I have a new EPOC+ with a manufactured date of Jan 03, 2016.  With some tweaking I managed to get the headset to start working in the current branch of Emokit.  However, the values coming out of the kit do not seem right.  I have no electrodes attached to the headset, however the readings are extremely varied and I am not quite sure what I am looking at.

+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |   2068   |   753    |    Okay    |  Nothing   |
|   FC5  |  -5321   |  11677   |    None    |  Excellent  |
|   AF3  |  -6884   |  16269   |    None    |  Excellent  |
|   F7   |  -7062   |   7162   |    None    |  Excellent  |
|   T7   |  -2848   |   9569   |    None    |  Excellent  |
|   P7   |  -3045   |   9731   |    None    |  Excellent  |
|   O1   |   4394   |   4742   |    None    |    None    |
|   O2   |   519    |   8387   |    None    |    None    |
|   P8   |   2643   |  12962   |    None    |  Excellent  |
|   T8   |   3402   |   2828   |    None    |    None    |
|   F8   |   3949   |   108    |  Nothing   |  Nothing   |
|   AF4  |   139    |   2870   |    None    |    None    |
|   FC6  |   -527   |  14005   |    None    |  Excellent  |
|   F4   |  -3407   |  14177   |    None    |  Excellent  |
|   X    |    63    |   N/A    |    N/A     |    N/A     |
|   Y    |   -10    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |   227    |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  4775    |  Packets Processed:  4774  |
|   Sampling Rate:   192    |        Crypto Rate:   192  |
+========================================================+

I also had to hard code the battery value to 0 as it kept tossing up values that were not found in the battery.py.

I have looked at emofox and emokit fork and need to check the numbers with the EEG software. emofox being the more interesting since the emokit fork does not have any writing out.  I have a 1 month subscription to the Premium SDK and I may be able to extend it to 2 to get this all working.

The headset I have is have set the sampling rate to 256, and as you can see above, I am not quite making it to 256, however, it holds steady around 192.

Another thing to note for those using USB 3.0, I have found (atleast on my Win7 Laptop) that it introduces alot of dropped packets on the pure.eeg software, where the good old 2.0 did not.

And lastly, I have a moderate (a bit Webserver based) programing skillset, and I have been looking at the .edf standard.  Once I figure out the formating, I was looking to write a python to output in .edf to be used in other software.  If I couldn't write it in python there would be a web applet you could load emotiv csv into and get back an edf at the very least.

bschumacher, 2017-04-03:
Probably need to set is_research=True

bschumacher, 2017-04-03:
A semi-good test, if the headset is not moving, the x and y values should be 0.

minifiredragon, 2017-04-04:
I have the is_research set to true (also tried false).  X and Y are all over the place as well.

minifiredragon, 2017-04-04:
I did some messing around with the emofox creation of emokit. In the code there was something set as a packet loop counter, and I noticed it was set to 127 (which I assume is 128 sample rate of the epoc) I have my headset (and it came out of the box preset) set to 256 samples.  So I changed that number to 255 and the timestamp error the unit was reporting disappeared and the battery indicator went to 0 instead of jumping around (If I read correctly, the battery is not always reported).  The senors (atleast from what I can tell) did not steady out (X and Y were still moving), but I think this is a step in the right direction.

I was trying to find a similar function within emokit, but I could not find it, but as I noted before my sampling was at 192 on the emokit test.  Perhaps if I could find the loop counter I can adjust I can get the battery error in this build to goto 0 as well?

bschumacher, 2017-04-04:
The counter is the first value in the packet data.
Set verbose=False, display_output=False and where we have this in packet.py, ~line 30:

            data = self.raw_data
            self.counter = data[0]

Add:
            data = self.raw_data
            self.counter = data[0]
            print(self.counter)

You should see it looping.

minifiredragon, 2017-04-05:
Ok, looking at the output it goes from 0 to 255.  I changed the if self.counter > 127 to 255 and the battery error stopped and the battery now reads 0 all the time.  I am not sure what this means exactly, I am assuming the sensor outputs have been remapped to different byte addresses or mixed in the the 9 new senor outputs.

bschumacher, 2017-04-05:
What about the X and Y?

bschumacher, 2017-04-05:
The battery level is calculated from the counter value, by changing the if statement you are essentially telling it to never calculate the battery level until the counter is greater than 255, which will not happen.

bschumacher, 2017-04-05:
So, the battery level is calculated in some other way now and probably everything else.

For a quick fix I would see if you can change the sampling rate to 128hz.

If you could though, post the raw data with the 256hz setting.

Set write=True and write_decrypted=True, it should generate the file.

bschumacher, 2017-04-05:
The good news, decryption is working it seems.

minifiredragon, 2017-04-05:
I tried all 3 settings of the Epoch+, which is Epoch, Epoch+ 128hz, Epoch+ 256.  Each one gave the same battery error.

I will try it with the battery forced to 0 and post all three setting results.

minifiredragon, 2017-04-05:
Here is a dump from the headset set to Epoch:
+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |   2290   |    0     |  Nothing   |  Nothing   |
|   FC5  |  -7542   |   824    |    Okay    |  Nothing   |
|   AF3  |  -8102   |   8147   |    None    |  Excellent  |
|   F7   |   3140   |  13343   |    None    |  Excellent  |
|   T7   |   1542   |   7717   |    None    |  Excellent  |
|   P7   |  -1799   |  15694   |    None    |  Excellent  |
|   O1   |  -3397   |  15238   |    None    |  Excellent  |
|   O2   |  -1425   |    0     |  Nothing   |  Excellent  |
|   P8   |   501    |    82    |  Nothing   |  Nothing   |
|   T8   |   5108   |  11623   |    None    |  Excellent  |
|   F8   |   5100   |  10615   |    None    |  Excellent  |
|   AF4  |  -3132   |  11482   |    None    |  Excellent  |
|   FC6  |  -4067   |   5117   |    None    |    None    |
|   F4   |  -1776   |   2474   |    None    |    None    |
|   X    |   -19    |   N/A    |    N/A     |    N/A     |
|   Y    |   -92    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:   632    |  Packets Processed:   631  |
|   Sampling Rate:   127    |        Crypto Rate:   127  |
+========================================================+

writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
[... 17 lines cut ...]
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings

[emotiv_values_2017-04-04 23-39-47.166997.txt](https://github.com/openyou/emokit/files/895126/emotiv_values_2017-04-04.23-39-47.166997.txt)
[epoch epoch emotiv_data_2017-04-04 23-39-47.166807.txt](https://github.com/openyou/emokit/files/895128/epoch.epoch.emotiv_data_2017-04-04.23-39-47.166807.txt)

minifiredragon, 2017-04-05:
This was the headset set for Epoc+ 128hz 16bit with MEMS set to 128hz 16bit

[epoch + 128hz emotiv_data_2017-04-04 23-50-58.624976.txt](https://github.com/openyou/emokit/files/895138/epoch.128hz.emotiv_data_2017-04-04.23-50-58.624976.txt)
[epoch + 128hz emotiv_values_2017-04-04 23-50-58.625369.txt](https://github.com/openyou/emokit/files/895139/epoch.128hz.emotiv_values_2017-04-04.23-50-58.625369.txt)

+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |   6381   |   8096   |    None    |  Excellent  |
|   FC5  |   4748   |   2649   |    None    |    None    |
|   AF3  |  -3529   |  12084   |    None    |  Excellent  |
|   F7   |   3296   |   3316   |    None    |    None    |
|   T7   |   2729   |  14318   |    None    |  Excellent  |
|   P7   |  -5698   |  14387   |    None    |  Excellent  |
|   O1   |   1263   |   6065   |    None    |  Excellent  |
|   O2   |   1305   |   9248   |    None    |  Excellent  |
|   P8   |   921    |   998    |    Okay    |  Nothing   |
|   T8   |   193    |  14953   |    None    |  Excellent  |
|   F8   |   6932   |  10499   |    None    |  Excellent  |
|   AF4  |  -2434   |   1987   |    None    |    None    |
|   FC6  |  -1936   |   9487   |    None    |  Excellent  |
|   F4   |  -1967   |   173    |  Nothing   |  Nothing   |
|   X    |   -85    |   N/A    |    N/A     |    N/A     |
|   Y    |   111    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  1274    |  Packets Processed:  1274  |
|   Sampling Rate:   258    |        Crypto Rate:   259  |
+========================================================+
writelines() argument must be a sequence of strings

writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
.... same sentence over and over before it redisplay the above chart.

minifiredragon, 2017-04-05:
All Tests are done with the headset in a box, and the felt pads attached and salined.
This was the headset set for Epoc+ 256hz 16bit with MEMS set to 128hz 16bit

[epoch+ 256hz 16bit MEMS 128hz 16bit emotiv_data_2017-04-04 23-58-20.408390.txt](https://github.com/openyou/emokit/files/895145/epoch.256hz.16bit.MEMS.128hz.16bit.emotiv_data_2017-04-04.23-58-20.408390.txt)
[epoch+ 256hz 16bit MEMS 128hz 16bit emotiv_values_2017-04-04 23-58-20.408835.txt](https://github.com/openyou/emokit/files/895146/epoch.256hz.16bit.MEMS.128hz.16bit.emotiv_values_2017-04-04.23-58-20.408835.txt)

+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |  -7273   |   6816   |    None    |  Excellent  |
|   FC5  |   4341   |  13106   |    None    |  Excellent  |
|   AF3  |   1487   |   2621   |    None    |    None    |
|   F7   |   1240   |  14216   |    None    |  Excellent  |
|   T7   |  -4065   |   8238   |    None    |  Excellent  |
|   P7   |   8187   |   6894   |    None    |  Excellent  |
|   O1   |   6821   |   3606   |    None    |    None    |
|   O2   |   730    |   7904   |    None    |    None    |
|   P8   |  -7140   |   6792   |    None    |  Excellent  |
|   T8   |   -640   |  11476   |    None    |  Excellent  |
|   F8   |   8145   |   9976   |    None    |  Excellent  |
|   AF4  |  -4021   |  12469   |    None    |  Excellent  |
|   FC6  |  -6595   |   6780   |    None    |  Excellent  |
|   F4   |  -1276   |   5568   |    None    |  Excellent  |
|   X    |   -19    |   N/A    |    N/A     |    N/A     |
|   Y    |    44    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  4708    |  Packets Processed:  4707  |
|   Sampling Rate:   396    |        Crypto Rate:   396  |
+========================================================+

writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
writelines() argument must be a sequence of strings
.... line repeats over and over until it redisplays the above chart.

bschumacher, 2017-04-05:
Alright, looks like we got a bug, I'll get that fixed so we can proceed.

bschumacher, 2017-04-05:
Bug should be squashed, hopefully.

If you could pull the latest master, I included a export decrypted example file.

minifiredragon, 2017-04-05:
Did you want me to do all three tests or is this one sufficient?

This was the headset set for Epoc+ 256hz 16bit with MEMS set to 128hz 16bit - No pads attached, just sitting in a box.

[emotiv_data_2017-04-05 12-12-35.446223.txt](https://github.com/openyou/emokit/files/900067/emotiv_data_2017-04-05.12-12-35.446223.txt)
[emotiv_values_2017-04-05 12-12-35.446426.txt](https://github.com/openyou/emokit/files/900068/emotiv_values_2017-04-05.12-12-35.446426.txt)

Emokit - v0.0.8 SN: UD20160103001874  Old Model: False
+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |   -464   |   2240   |    None    |    None    |
|   FC5  |   2610   |   9687   |    None    |  Excellent  |
|   AF3  |   4586   |   6141   |    None    |  Excellent  |
|   F7   |  -6576   |   5372   |    None    |  Excellent  |
|   T7   |   7901   |  13419   |    None    |  Excellent  |
|   P7   |   -979   |  15473   |    None    |  Excellent  |
|   O1   |   -741   |   3359   |    None    |    None    |
|   O2   |  -5713   |   7507   |    None    |    None    |
|   P8   |  -3234   |  14883   |    None    |  Excellent  |
|   T8   |  -7410   |  10869   |    None    |  Excellent  |
|   F8   |   6512   |   7300   |    None    |  Excellent  |
|   AF4  |   3050   |  14716   |    None    |  Excellent  |
|   FC6  |  -1658   |  16318   |    None    |  Excellent  |
|   F4   |   2847   |   1834   |    None    |    None    |
|   X    |    73    |   N/A    |    N/A     |    N/A     |
|   Y    |   -23    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  5847    |  Packets Processed:  5846  |
|   Sampling Rate:   392    |        Crypto Rate:   391  |
+========================================================+

bschumacher, 2017-04-05:
The counter does not appear to be accurate can you toggle the is_research?

minifiredragon, 2017-04-05:
Emotiv(display_output=True, verbose=True, write=True, write_decrypted=True, is_research=True) as headset: is the setting for the above run.  

Here is it with the setting to False:

[emotiv_data_2017-04-05 12-20-55.827315.txt](https://github.com/openyou/emokit/files/900084/emotiv_data_2017-04-05.12-20-55.827315.txt)
[emotiv_values_2017-04-05 12-20-55.827542.txt](https://github.com/openyou/emokit/files/900085/emotiv_values_2017-04-05.12-20-55.827542.txt)

Emokit - v0.0.8 SN: UD20160103001874  Old Model: False
+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |  -6170   |  12589   |    None    |  Excellent  |
|   FC5  |  -6260   |  14510   |    None    |  Excellent  |
|   AF3  |   3825   |  13400   |    None    |  Excellent  |
|   F7   |   735    |   4758   |    None    |    None    |
|   T7   |   4069   |   5194   |    None    |  Excellent  |
|   P7   |  -6221   |   9625   |    None    |  Excellent  |
|   O1   |   292    |   8173   |    None    |  Excellent  |
|   O2   |  -6846   |   2818   |    None    |  Excellent  |
|   P8   |  -6337   |   9274   |    None    |  Excellent  |
|   T8   |   7214   |   4353   |    None    |    None    |
|   F8   |  -7949   |   401    |  Nothing   |  Nothing   |
|   AF4  |  -2944   |   8614   |    None    |  Excellent  |
|   FC6  |  -4257   |   8220   |    None    |  Excellent  |
|   F4   |   2641   |   5763   |    None    |  Excellent  |
|   X    |   147    |   N/A    |    N/A     |    N/A     |
|   Y    |    94    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  5844    |  Packets Processed:  5844  |
|   Sampling Rate:   391    |        Crypto Rate:   392  |
+========================================================+

minifiredragon, 2017-04-05:
Also, just tried to run your encrypted example, it said this:

sequence item 1: expected string, list found

bschumacher, 2017-04-05:
Would you included files from the other modes as well?

minifiredragon, 2017-04-05:
Did you want me to do all three tests or is this one sufficient?

This was the headset set for Epoc+ 256hz 16bit with MEMS set to 128hz 16bit - No pads attached, just sitting in a box.
is_research=True

[emotiv_data_2017-04-05 12-34-12.865869.txt](https://github.com/openyou/emokit/files/900115/emotiv_data_2017-04-05.12-34-12.865869.txt)
[emotiv_values_2017-04-05 12-34-12.866069.txt](https://github.com/openyou/emokit/files/900116/emotiv_values_2017-04-05.12-34-12.866069.txt)
Emokit - v0.0.8 SN: UD20160103001874  Old Model: False
+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |   354    |   499    |  Nothing   |  Nothing   |
|   FC5  |  -6667   |   1643   |    None    |    None    |
|   AF3  |   2466   |   1507   |    Good    |    None    |
|   F7   |  -2674   |  13333   |    None    |  Excellent  |
|   T7   |   1223   |  14432   |    None    |  Excellent  |
|   P7   |   -167   |  14102   |    None    |  Excellent  |
|   O1   |   2224   |   1999   |    None    |    None    |
|   O2   |  -7023   |   732    |    Okay    |    None    |
|   P8   |   3788   |   6102   |    None    |  Excellent  |
|   T8   |   5414   |   295    |  Nothing   |  Nothing   |
|   F8   |  -6007   |  15468   |    None    |  Excellent  |
|   AF4  |   5005   |  12238   |    None    |  Excellent  |
|   FC6  |  -5084   |   8255   |    None    |  Excellent  |
|   F4   |  -6255   |   8063   |    None    |  Excellent  |
|   X    |   -97    |   N/A    |    N/A     |    N/A     |
|   Y    |    40    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  5844    |  Packets Processed:  5843  |
|   Sampling Rate:   393    |        Crypto Rate:   393  |
+========================================================+

bschumacher, 2017-04-05:
Do that one again with is_research=False please.

minifiredragon, 2017-04-05:
This was the headset set for Epoc+ 128hz 16bit with MEMS set to 128hz 16bit - No pads attached, just sitting in a box.
is_research=False

[emotiv_data_2017-04-05 12-38-07.343700.txt](https://github.com/openyou/emokit/files/900129/emotiv_data_2017-04-05.12-38-07.343700.txt)
[emotiv_values_2017-04-05 12-38-07.343961.txt](https://github.com/openyou/emokit/files/900130/emotiv_values_2017-04-05.12-38-07.343961.txt)

Emokit - v0.0.8 SN: UD20160103001874  Old Model: False
+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |  -6531   |  11121   |    None    |  Excellent  |
|   FC5  |   3812   |  14533   |    None    |  Excellent  |
|   AF3  |  -2154   |   5746   |    None    |  Excellent  |
|   F7   |   4629   |   2782   |    None    |    None    |
|   T7   |   7267   |   3138   |    None    |    None    |
|   P7   |   254    |   6611   |    None    |  Excellent  |
|   O1   |   7419   |  10204   |    None    |  Excellent  |
|   O2   |   -130   |   6561   |    None    |  Excellent  |
|   P8   |  -7155   |  14440   |    None    |  Excellent  |
|   T8   |   5927   |  11274   |    None    |  Excellent  |
|   F8   |  -3077   |   7347   |    None    |  Excellent  |
|   AF4  |  -1372   |  12847   |    None    |  Excellent  |
|   FC6  |  -1739   |   2835   |    None    |    None    |
|   F4   |   -788   |   6388   |    None    |  Excellent  |
|   X    |    87    |   N/A    |    N/A     |    N/A     |
|   Y    |   146    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  2314    |  Packets Processed:  2313  |
|   Sampling Rate:   390    |        Crypto Rate:   390  |
+========================================================+

minifiredragon, 2017-04-05:
This was the headset set for Epoc 128hz 14bit (No MEMS) - No pads attached, just sitting in a box.
is_research=True

[emotiv_data_2017-04-05 12-42-01.571956.txt](https://github.com/openyou/emokit/files/900139/emotiv_data_2017-04-05.12-42-01.571956.txt)
[emotiv_values_2017-04-05 12-42-01.572241.txt](https://github.com/openyou/emokit/files/900140/emotiv_values_2017-04-05.12-42-01.572241.txt)

+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |   2080   |   9155   |    None    |  Excellent  |
|   FC5  |   1678   |  14761   |    None    |  Excellent  |
|   AF3  |   968    |   7857   |    None    |  Excellent  |
|   F7   |  -1121   |   5024   |    None    |    None    |
|   T7   |  -3002   |   7433   |    None    |  Excellent  |
|   P7   |   1115   |  15043   |    None    |  Excellent  |
|   O1   |  -5262   |  10140   |    None    |  Excellent  |
|   O2   |  -1558   |  15473   |    None    |  Excellent  |
|   P8   |   240    |   3817   |    None    |    None    |
|   T8   |   693    |   1077   |    Okay    |    None    |
|   F8   |   6417   |   7117   |    None    |  Excellent  |
|   AF4  |   4851   |  11588   |    None    |  Excellent  |
|   FC6  |   3939   |   3838   |    None    |    None    |
|   F4   |  -2100   |  15285   |    None    |  Excellent  |
|   X    |   -71    |   N/A    |    N/A     |    N/A     |
|   Y    |   -38    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  3096    |  Packets Processed:  3095  |
|   Sampling Rate:   390    |        Crypto Rate:   390  |
+========================================================+

minifiredragon, 2017-04-05:
This was the headset set for Epoc 128hz 14bit (No MEMS) - No pads attached, just sitting in a box.
is_research=False

[emotiv_data_2017-04-05 12-44-14.570546.txt](https://github.com/openyou/emokit/files/900147/emotiv_data_2017-04-05.12-44-14.570546.txt)
[emotiv_values_2017-04-05 12-44-14.570748.txt](https://github.com/openyou/emokit/files/900148/emotiv_values_2017-04-05.12-44-14.570748.txt)

+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |  -2558   |  12064   |    None    |  Excellent  |
|   FC5  |  -6184   |  15801   |    None    |  Excellent  |
|   AF3  |  -3255   |   2516   |    None    |    None    |
|   F7   |   7698   |   136    |  Nothing   |  Nothing   |
|   T7   |   334    |   7484   |    None    |  Excellent  |
|   P7   |   2456   |   8502   |    None    |  Excellent  |
|   O1   |  -1764   |   2437   |    None    |    None    |
|   O2   |  -5047   |  15901   |    None    |    None    |
|   P8   |   1916   |  13338   |    None    |  Excellent  |
|   T8   |  -7028   |  13222   |    None    |  Excellent  |
|   F8   |   6568   |   7732   |    None    |  Excellent  |
|   AF4  |   2226   |   7567   |    None    |  Excellent  |
|   FC6  |   -750   |  15479   |    None    |  Excellent  |
|   F4   |   6191   |   2378   |    None    |    None    |
|   X    |   109    |   N/A    |    N/A     |    N/A     |
|   Y    |    54    |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  3889    |  Packets Processed:  3888  |
|   Sampling Rate:   394    |        Crypto Rate:   394  |
+========================================================+

minifiredragon, 2017-04-05:
I forgot to note that I had to edit packet.py and force self.battery = 0 for the file to run.

bschumacher, 2017-04-05:
What is the error? None of these show the counter looping.

You said earlier that it was looping though?

minifiredragon, 2017-04-05:
Battery error:

   AES KEY - [u'4', '\x00', u'7', 'T', u'8', '\x10', u'1', 'B', u'4', '\x00', u'7', 'H', u'8', '\x00', u'1', 'P']
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 811, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 764, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/minifiredragon/EEG/emokit-master/python/emokit/emotiv.py", line 298, in run
    new_packet = EmotivPacket(decrypted_task.data, timestamp=decrypted_task.timestamp)
  File "/home/minifiredragon/EEG/emokit-master/python/emokit/packet.py", line 40, in __init__
    self.battery = battery_values[str(self.counter)]
KeyError: '210'

When you had me add the extra print line, what was printing was like (not actual data):
34
244
0
45
255
1
45
78
43
245
9

bschumacher, 2017-04-05:
In Epoc mode we should see output similar to this:

[image]

bschumacher, 2017-04-05:
Ah yeah, perhaps decryption changed then.

minifiredragon, 2017-04-05:
With everything you have had me do, that never displayed on the screen. However, last night when I was messing around looking for the writeline() error, I did manage to get something similar, it started with [ and had a bunch of data with a /n at the end, an occasional expect string error message and after a bit had a ]  and started again with a [

minifiredragon, 2017-04-05:
What I have been posting is copy and pasted from my terminal window on my linux test box.

minifiredragon, 2017-04-05:
Also, the encyrption example you have give an error message along the lines of expected string and got list.

bschumacher, 2017-04-05:
Yeah, can you pull the latest master and post the output of hid_info.py

minifiredragon, 2017-04-05:
sys.version_info(major=2, minor=7, micro=5, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0015:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0015:01
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 50475
interface_number, 0
manufacturer_string, Logitech
vendor_id, 1133
release_number, 4611
serial_number, None
usage, 0
path, 0002:0009:00
product_string, USB Receiver
-------------------------
usage_page, 0
product_id, 50475
interface_number, 1
manufacturer_string, Logitech
vendor_id, 1133
release_number, 4611
serial_number, None
usage, 0
path, 0002:0009:01
product_string, USB Receiver
-------------------------
usage_page, 0
product_id, 50475
interface_number, 2
manufacturer_string, Logitech
vendor_id, 1133
release_number, 4611
serial_number, None
usage, 0
path, 0002:0009:02
product_string, USB Receiver
-------------------------
usage_page, 0
product_id, 8198
interface_number, 0
manufacturer_string, ELAN
vendor_id, 1267
release_number, 4369
serial_number, None
usage, 0
path, 0002:0002:00
product_string, Touchscreen
************************************************************
! Please include this information if you open a new issue. !
************************************************************

bschumacher, 2017-04-05:
Thanks, ok so with the latest master you should be able to write encrypted.

bschumacher, 2017-04-05:
Also, if you change the mode on the headset does the hid_info.py output change?

minifiredragon, 2017-04-05:
I am running out for a little.  when I return I will post the encrypted data as well as check to see if the hid_info changes when I change the headset.

bschumacher, 2017-04-05:
Great, we'll get this thing figured out ;)

minifiredragon, 2017-04-05:
Here is the encryped version:
This was the headset set for Epoc+ 256hz 16bit with MEMS set to 128hz 16bit - No pads attached, just sitting in a box.
is_research=True

[emotiv_encrypted_data_UD20160103001874_2017-04-05 17-21-32.384061.txt](https://github.com/openyou/emokit/files/900828/emotiv_encrypted_data_UD20160103001874_2017-04-05.17-21-32.384061.txt)
[emotiv_values_2017-04-05 17-21-32.384281.txt](https://github.com/openyou/emokit/files/900829/emotiv_values_2017-04-05.17-21-32.384281.txt)

*****----- HID_INFO IS CUT OFF, ONLY SHOWING Emotiv -------------*****
sys.version_info(major=2, minor=7, micro=5, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0016:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0

sys.version_info(major=2, minor=7, micro=5, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0016:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0016:01
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------

path, 0001:0016:01
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------

minifiredragon, 2017-04-05:
Here is the encryped version:
This was the headset set for Epoc+ 256hz 16bit with MEMS set to 128hz 16bit - No pads attached, just sitting in a box.
is_research=False

[emotiv_encrypted_data_UD20160103001874_2017-04-05 17-24-56.898471.txt](https://github.com/openyou/emokit/files/900838/emotiv_encrypted_data_UD20160103001874_2017-04-05.17-24-56.898471.txt)
[emotiv_values_2017-04-05 17-24-56.898709.txt](https://github.com/openyou/emokit/files/900839/emotiv_values_2017-04-05.17-24-56.898709.txt)

----- HID_INFO IS CUT OFF, ONLY SHOWING Emotiv -------------
sys.version_info(major=2, minor=7, micro=5, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0016:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0016:01
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------

minifiredragon, 2017-04-05:
Here is the encryped version:
This was the headset set for Epoc+ 128hz 16bit with MEMS set to 128hz 16bit - No pads attached, just sitting in a box.
is_research=True

[emotiv_encrypted_data_UD20160103001874_2017-04-05 17-30-51.246114.txt](https://github.com/openyou/emokit/files/900850/emotiv_encrypted_data_UD20160103001874_2017-04-05.17-30-51.246114.txt)
[emotiv_values_2017-04-05 17-30-51.246369.txt](https://github.com/openyou/emokit/files/900852/emotiv_values_2017-04-05.17-30-51.246369.txt)

----- HID_INFO IS CUT OFF, ONLY SHOWING Emotiv -------------
sys.version_info(major=2, minor=7, micro=5, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0017:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0017:01
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------

minifiredragon, 2017-04-05:
Here is the encryped version:
This was the headset set for Epoc+ 128hz 16bit with MEMS set to 128hz 16bit - No pads attached, just sitting in a box.
is_research=False

[emotiv_encrypted_data_UD20160103001874_2017-04-05 17-33-32.632431.txt](https://github.com/openyou/emokit/files/900855/emotiv_encrypted_data_UD20160103001874_2017-04-05.17-33-32.632431.txt)
[emotiv_values_2017-04-05 17-33-32.632676.txt](https://github.com/openyou/emokit/files/900857/emotiv_values_2017-04-05.17-33-32.632676.txt)

----- HID_INFO IS CUT OFF, ONLY SHOWING Emotiv -------------

sys.version_info(major=2, minor=7, micro=5, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0017:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0017:01
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------

minifiredragon, 2017-04-05:
Here is the encryped version:
This was the headset set for Epoc 128hz 14bit no MEMS  - No pads attached, just sitting in a box.
is_research=True

[emotiv_encrypted_data_UD20160103001874_2017-04-05 17-39-48.516489.txt](https://github.com/openyou/emokit/files/900872/emotiv_encrypted_data_UD20160103001874_2017-04-05.17-39-48.516489.txt)
[emotiv_values_2017-04-05 17-39-48.516766.txt](https://github.com/openyou/emokit/files/900873/emotiv_values_2017-04-05.17-39-48.516766.txt)

----- HID_INFO IS CUT OFF, ONLY SHOWING Emotiv -------------

sys.version_info(major=2, minor=7, micro=5, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0018:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0018:01
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------

minifiredragon, 2017-04-05:
Here is the encryped version:
This was the headset set for Epoc 128hz 14bit no MEMS - No pads attached, just sitting in a box.
is_research=False

[emotiv_encrypted_data_UD20160103001874_2017-04-05 17-42-23.292665.txt](https://github.com/openyou/emokit/files/900883/emotiv_encrypted_data_UD20160103001874_2017-04-05.17-42-23.292665.txt)
[emotiv_values_2017-04-05 17-42-23.293088.txt](https://github.com/openyou/emokit/files/900885/emotiv_values_2017-04-05.17-42-23.293088.txt)

----- HID_INFO IS CUT OFF, ONLY SHOWING Emotiv -------------

sys.version_info(major=2, minor=7, micro=5, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 60674
interface_number, 0
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0018:00
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------
usage_page, 0
product_id, 60674
interface_number, 1
manufacturer_string, Emotiv
vendor_id, 4660
release_number, 6
serial_number, UD20160103001874
usage, 0
path, 0001:0018:01
product_string, Brain Computer Interface USB Receiver/Dongle
-------------------------

bschumacher, 2017-04-05:
Ok, thanks for all the data. I will now attempt to create an encryption key solver.

I have no idea how long it will take, but since we know the data format for the normal Epoc, we have the encrypted data and a working headset/key it should be fairly easy and dare I say... fun. lol

minifiredragon, 2017-04-05:
Always fun.  I wonder what I did with my gpu powered crypto breaking software I had bought years ago. It let you put in a salt and it would run until it was solved.  I will have to see if I can dig that up somewhere.

minifiredragon, 2017-04-05:
Keep in mind I also have pure.eeg as I have also bought the subscription which will return headset data in edf (though not a valid edf) and CSV.

bschumacher, 2017-04-05:
Nice, once I/we figure out this encryption we're definitely gonna put that to good use.

bschumacher, 2017-04-06:
Alright, so I started on the solver. It hasn't found it yet, but might eventually.

I'm going to add brute force and multiprocessing later, feel free to hack on it, it is in the latest master.

bschumacher, 2017-04-06:
The counter check piece of it is good, made sure with my headset.

minifiredragon, 2017-04-06:
The counter, that is print(self.counter) in packet.  How is it suppose to look coming out?

bschumacher, 2017-04-06:
It should increment 0-127, some value over 128 and repeat.

minifiredragon, 2017-04-06:
I edit the above post as u responded.

minifiredragon, 2017-04-06:
The above is is_research=True and the False gives the same output

minifiredragon, 2017-04-06:
Need the text file for the key_solver

bschumacher, 2017-04-06:
Using the Epoc setting? I suspect that they are keying off of the manufacture date for the key change. Which is precisely what I'm going to do once I figure it out.

The file is the same ones you sent me.

minifiredragon, 2017-04-06:
I just noticed that.  I was deleting them off my pc so I wasn't paying attention to what was in the emokit-master.  I got it back.  

The headset is in the epoc+ 256hz mode.  I will look at it at the epoc mode.

minifiredragon, 2017-04-06:
Headset set to epoc 128hz 14bit

Initializing Emokit...
Initializing Reader Thread...
Initializing Writer Thread(s)...
Initializing Crypto Thread...
Initializing Output Thread...
86
250
6
208
55
4
224
140
208
47
17
254
171
18
68
27
116
201
187
90
15
138
17
22
190
172
253
154
157
214
179
122

[312 more replies]

closed 2017-04-14 by bschumacher