openyou/emokit #251: Corrupted output or `Could not open device.` on Mac OSX.

matthew-groth, 2017-04-13:
**Summary**

I am using OSX: 10.12.4 with an EPOC 1.0 and the latest version of this repo. The EPOC works fine with Emotiv's software. With an old version of emokit (about a year ago), we were able to get it working _on _linux__ but not mac. I am giving the mac another attempt now with the updated repo, but it usually does not work. 

**Outputs**

Every time I run Example.py, I get 1 of 3 possible situations. Click to download a copy of the full output.
1- [RuntimeError: Could not open device](https://github.com/openyou/emokit/files/919683/Could.Not.Open.Device.txt)
2- [Corrupted output stream](https://github.com/openyou/emokit/files/919681/Corrupted.Output.txt)
3- It works. This is very rare, but [I managed to reproduce it for this issue](https://github.com/openyou/emokit/files/919684/It.Works.txt).

Each output file has the command I ran at the top: `sudo python example.py`, followed by the full output until I canceled with ctrl-z. I'm printing a bit of extra info from print_hid_enumerate(). Let me know if there's anything else I should be looking for or printing because I'm unfamiliar with all of this.

**Patterns**

In general if it doesn't work the first time I have to take the dongle out and put it back in, but there has been one or two times where it worked without me doing that. I think the corrupted data happens more if I try to run soon after the connection is established, where as `could not open` occurs if I wait a bit. I also think it might be more likely to work if I wait, but most of the time this leads to a `could not open`.

bschumacher, 2017-04-13:
The key doesn't look right, I think you might be a commit or two behind.

matthew-groth, 2017-04-13:
I'm not sure what you mean by key, and I thought I forked recently. But I did a fresh fork and now it seems to be working a lot better. Closing for now, thanks.

closed 2017-04-13 by matthew-groth

bschumacher, 2017-04-13:
Would you mind uploading an example_export_encrypted.py?