openyou/emokit #259: Emokit-osc Trace/BPT trap: 5

phonotribe, 2017-05-29:
I am using emokit-osc for a year now and everything worked great. Yesterday I tried to run the emokit-osc and I am getting the following error:

```
dyld: Symbol not found: __ZTI9UdpSocket
  Referenced from: /Users/Kyriakos/Downloads/emokit-c-master/bin/emokit_osc
  Expected in: /usr/local/lib/liboscpack.so
 in /Users/Kyriakos/Downloads/emokit-c-master/bin/emokit_osc
Trace/BPT trap: 5
logout
```
It seems like a linking problem but I didnt find any way to worked it around.
Any ideas?

warrenarea, 2017-05-31:
http://osc_dev.create.ucsb.narkive.com/P0LsdTCT/oscpack

From the sounds of it, there is a linkage problem, looks like they suggest 

```
Turn off the zero link flag/option in the build configuration and
recompile every thing again.
```

or

```
Looks like you need to link with the right symbols. That would be liboscpack if you've made a static lib, or else just include the source files in your project (what I usually do)..

```

phonotribe, 2017-06-02:
Thanks warrenarea, I recompiled the oscpack and that solved the problem.

closed 2017-06-02 by phonotribe