openyou/emokit #260: File name for data collected .csv

nadunindunil, 2017-06-05:
It seems emokit doesn't provide a way to create a .csv file with a name of the user desired. wouldn't it be great to have a way to rename it?

bschumacher, 2017-06-05:
This was intentional as the file name is used to playback the recordings without having the headset plugged in or specifying the serial number in addition to the method in which to read the file, was it raw data, encrypted data or sensor values for example.

However, the prefix could probably be used as a custom identifier or something.. or maybe determining the read based off the data or header in the file itself. Maybe an additional header?

If you have a good solution feel free to submit a PR.

nadunindunil, 2017-06-05:
@bschumacher thank you very much for the quick reply. sure I have few ideas and I 'll submit a PR as soon as I finished them.

closed 2017-06-05 by nadunindunil