openyou/emokit #261: How could I get sensors value array

shihray, 2017-06-14:
hi everyone
i'm try to use headset.dequeue() to get values
gust like packet.sensors['AF3']['value']
but this just can get one value
how to get sensors value to array

bschumacher, 2017-06-14:
packet.sensors is a dictionary of all the values.

shihray, 2017-06-14:
@bschumacher thanks for your answer
but I want to get an array its length is 128 in one second
how could I make that come true

bschumacher, 2017-06-14:
You would append the packet.sensors to a list, unless you're using numpy arrays.

When the list length is equal to 128 append that list to another list assuming you want to keep it around or perform your processing or whatever.

shihray, 2017-06-14:
@bschumacher thanks a lot 
I'm got it

closed 2017-06-14 by bschumacher