openyou/emokit #263: Which headset to buy?

warrenarea, 2017-06-14:
I think I've finally made a decision, that when it goes on its 20% sale, i'm going to go with the Insight.

I would really like the Epoc+ but even at discount its $640

looking at my other options, even OpenBCI, to get a similar 14 channels, you'd end up spending 
at least a thousand.  Emotiv (besides for the licensing nonsense) is still priced pretty well.

And if you look at the comparison chart, Insight can measure "just about" the same exact reactions as
the Epoc+with a few exceptions of a smirk, which i can do without. 

The extra measurement locations is nice, but at the same time.... i have to evaluate, that there is only
so much that this specific technology can do.   I would eventually like to maybe explore using different
algorithms and such and explore different measuring techniques.

Plus, I don't want to take a chance on the Epoc+ plastic... though they claim the polycarbonate is more 
pure now... and Insight more than likely uses the same plastics, but the frame seems a little bit sturdier,
and I cannot deny that the polymer gel contacts would make developing for it 10x easier. 

also factoring in that this is a consumer model, that many others will have, is good for developing for
it as well.  It may also prove handy as well... as none of us have an Insight.

Though even with their 20% discount, I still have to buy the dongle for it, so at its $299 price tag, it
will still end up costing me $288  (with a $60 dongle at 20%), so more or less, i'll be saving $10 or
$60 depending on how you look at it.

I did see some functions in IDA for some "cognitive detections", perhaps that could come in handy.

Though I think there might come a point, where we have reversed so much of their program, they 
might not take as kindly as they did in the past.

warrenarea, 2017-06-19:
I couldn't make up my mind on choosing one, so I got both of them.

Figured it may be a while before their 20% sale rolls around again, (probably in Fall and or Christmas)
and today was their last day.

Plus, I think I'm in a unique position to help others who have the headset, and will expedite the 
process of deciphering how it works... since I'll have every headset, i'll be able to see which parts
of code each headset is climbing into. 

Also, considering I was on the fence about the frailty of the plastics involved... and knowing what past
models were like, i'll be able to make a critique on whether its improved. 

and I really can't wait to try the polymer contacts... I've been discussing them for so long, it'll be nice
to finally try them!

no doubt, it was a fair lump of change to lay down, but I think my interest and passion for this will
make it worth it.   Plus I have had some neat ideas on utilizing it, that couldn't be made possible
with my modified Epoc version.  May consider remaking that as well, if i can find the time.

warrenarea, 2017-06-19:
i will be interested to see how the dongles all work together...
probably could have figured it out on the emotiv forums.

Was curious whether the  universal dongle (for insight) will work with the Epoc+ and Epoc.

or whether the Epoc+ dongle (which comes included), is specifically married to the Epoc+
or could support other headsets.

and if the Epoc dongle would work with any of the others.

Maybe some good information for the FAQ section.

minifiredragon, 2017-06-28:
From what I read, it can be rekeyed to another headset.  Since you can buy extra dongles from the store, there would be no other way.  Also, when you plug the headset into the pc to change settings, while having the dongle in, a second (not usuable) headset appears.

warrenarea, 2017-06-28:
yeah I haven't even thought of looking into how it changes the Epoc+ settings.
In order for that to occur, i would think it'd "have" to have a back and forth communication
from the dongle to the headset.  

where the Epoc version, only had to be one-way communication.

And it can't just be sending "all" the data all the time, because that wouldn't mean it would be able to
conserve any power.

warrenarea, 2017-06-28:
scratch the wireless communication idea... it actually requires a physical USB connection in order to
change the settings from one mode to another...

In which case, I will want to look at the USB connection in emotiv.py
and then in IDA, find a HID command for writing/sending data.

closed 2017-08-02 by warrenarea

warrenarea, 2017-09-11:
Upon further review of the Epoc+ 

I'm not really impressed with the sensor contacts that plug into the arms of the device.

A couple of the contacts seem to fall out easily, this seems due to the engineering of the
plastic rings around the contacts, either being misshaped from manufacturing, or just 
breaking breaking from wear. 

I've only plugged in the sensors once, so the fact that I have issue with them so early,
is not a good sign.

They definitely need a new system for attaching the felt sensors. 

and I am still skeptical on the frame of the headset, it seems to be holding up, but i can't
see the stress of stretching the headset could last for very long.  I'm sure they make the
headset arch a little narrower so that it pushes the sensors onto your head... but without
having a more durable material... i am still cautious about it breaking.

warrenarea, 2017-09-11:
Upon further review of the insight.

Laying the headset down can be awkward, as it doesn't have the symmetry of the Epoc's and 
if you lay it down improperly, the device can flip over.

That might seem a little nit-picky, but i consider, it almost flipped over and fell off my desk.
and I keep expecting that I'm going to break the arms of the Insight just because of its 
unusual symmetry and long-reaching arms.

warrenarea, 2017-09-11:
Review of the Universal Dongle.
I may have posted about it already, but the universal dongle, does in fact detect
"all" of the devices.  Epoc, Epoc+ and Insight. 

Although it does pair to each device... you will often (if not always), have to unplug
the USB and plug it back in again, when getting it to pair to a new device.

warrenarea, 2017-09-11:
Imagine though, that schools probably buy these devices for research, or for their students. 
I could easily see many of these devices breaking on accident, especially with people who are
unfamiliar about how careful you have to handle these devices.

I learned a long time ago, you don't bring nice things to work, or to school, 
because they will always end up broken.