openyou/emokit #264: Key Code Update.

warrenarea, 2017-06-25:
```
        self.is_research = True
        model = 1
        k = ['\0'] * 16
        
        if model == 1:
            k[0] = sn[-1]
            k[1] = '\0'
            k[2] = sn[-2]
            k[3] = 'T'
            k[4] = sn[-3]
            k[5] = '\x10'
            k[6] = sn[-4]
            k[7] = 'B'
            k[8] = sn[-1]
            k[9] = '\0'
            k[10] = sn[-2]
            k[11] = 'H'
            k[12] = sn[-3]
            k[13] = '\0'
            k[14] = sn[-4]
            k[15] = 'P'
        
        if model == 2:
            k[0] = sn[-1]
            k[1] = '\0'
            k[2] = sn[-2]
            k[3] = 'H'
            k[4] = sn[-1]
            k[5] = '\0'
            k[6] = sn[-2]
            k[7] = 'T'
            k[8] = sn[-3]
            k[9] = '\x10'
            k[10] = sn[-4]
            k[11] = 'B'
            k[12] = sn[-3]
            k[13] = '\0'
            k[14] = sn[-4]
            k[15] = 'P'
         
        if model == 3:
            k[0] = sn[-2]
            k[1] = '\0'
            k[2] = sn[-1]
            k[3] = 'D'
            k[4] = sn[-2]
            k[5] = '\0'
            k[6] = sn[-1]
            k[7] = '\x0C'
            k[8] = sn[-4]
            k[9] = '\0'
            k[10] = sn[-3]
            k[11] = '\x15'
            k[12] = sn[-4]
            k[13] = '\0'
            k[14] = sn[-3]
            k[15] = 'X'
            
         
        if model == 4:
            k[0] = sn[-1]
            k[1] = '\0'
            k[2] = sn[-2]
            k[3] = '\x15'
            k[4] = sn[-3]
            k[5] = '\0'
            k[6] = sn[-4]
            k[7] = '\x0C'
            k[8] = sn[-3]
            k[9] = '\0'
            k[10] = sn[-2]
            k[11] = 'D'
            k[12] = sn[-1]
            k[13] = '\0'
            k[14] = sn[-2]
            k[15] = 'X'
            
         
        if model == 5:
            k[0] = sn[-2]
            k[1] = sn[-1]
            k[2] = sn[-2]
            k[3] = sn[-1]
            k[4] = sn[-3]
            k[5] = sn[-4]
            k[6] = sn[-3]
            k[7] = sn[-4]
            k[8] = sn[-4]
            k[9] = sn[-3]
            k[10] = sn[-4]
            k[11] = sn[-3]
            k[12] = sn[-1]
            k[13] = sn[-2]
            k[14] = sn[-1]
            k[15] = sn[-2]
          
        if model == 6:
            k[0] = sn[-1]
            k[1] = sn[-2]
            k[2] = sn[-2]
            k[3] = sn[-3]
            k[4] = sn[-3]
            k[5] = sn[-3]
            k[6] = sn[-2]
            k[7] = sn[-4]
            k[8] = sn[-1]
            k[9] = sn[-4]
            k[10] = sn[-2]
            k[11] = sn[-2]
            k[12] = sn[-4]
            k[13] = sn[-4]
            k[14] = sn[-2]
            k[15] = sn[-1]
            
                     
        
        
        key = ''.join(k)
```

put together a couple more keys for the epoc+
was going to put it in a push request, but I wasn't sure what arrangement you
wanted the code in, so i didn't figure I'd waste my time if you were going to change it anyways. lol

i considered using 'options' python command to execute different functions, but if its just for
setting up the key, its probably not worth it at this stage.

in the part where is_research is initialized, it might make more sense to just replace that with
model...

Model #1 is regular epoc
Model #2 is research_model
Model #3 Epoc+
Model #4 Epoc+ 
Model #5 Epoc+
Model #6 Epoc+ 

will provide more information on the Epoc+ keys/models soon.

Mateojs11, 2017-11-03:
hi, I have a problem, i try with all these keys, but they don't work,only the six key doesn't generate me error, but when plotting with render.py, the signals vary constantly. i have a Epoc + EEG.

[image]

you have more keys that I can try or some idea of how to solve the variation of signals, thank you very much.

warrenarea, 2017-11-07:
https://github.com/CymatiCorp/CyKITv2

Mateojs11, 2017-11-08:
hi, I try with CyKITv2 but the headset still don't work, the emotiv is a 2017 version, and de serial number is UD201701030026BD.

with the CyKITv2 I get the following graphics:

[image]
[image]
[image]
[image]

¿what else can Ido to advance? thanks for the help.

warrenarea, 2017-11-08:
basically, that is where i am at in the development process of the Epoc+

1st screenshot is all correct.

2nd screenshot, the baseline probably needs fixed,
 or i didn't completely fill out that feature. (it does work in epoc and insight)
 but even those, need a baseline fix.

3rd screenshot. that is the data... but is not deciphered properly.

4th screenshot, that is the counter data.... that is what its supposed to look like, and lets
                       us know it decrypted the packets.

Mateojs11, 2017-11-09:
Thank you very much for the answer, if the data is not properly deciphered means that the key is not correct? Is there a way to find the right key, maybe using the emokit's files  key_silver.py or key_solver_bruteforce ? I tried with key_solver_bruteforce_client and server and they ran but after an hour they did´t find a correct key.

warrenarea, 2017-11-09:
yeah i wasn't exactly clear on that.

you have the right key... because the counter will have that saw-tooth appearance.
thats the data going in sequential order from 0 - 255

its the data that needs filtered properly... basically for the Epoc version, that was what the
get_level function is for... Epoc+ currently isn't making use of any masking to sort out the
data.