openyou/emokit #265: Code for Epoc+ Mode Change.

warrenarea, 2017-07-01:
Finally figured out how to send a raw data packet to the USB. 
We now have a working mode changer for Epoc+

Have tested and verified the code works as well. 

the script i use requires you to confirm what device you want to use, 
so you may want to modify that part out.... or, have it ask what mode
the device should be put in.

change EPOC_ChangeMode to select what mode its to be put in.

```
elif device.product_name == 'EPOC+':
                    
                    print "\n" + device.product_name + " Found!\n"
                    useDevice = raw_input("Use this device? [Y]es? ")
                    if useDevice.upper() == "Y":                   
                         devicesUsed += 1
                         devices.append(device)
                         device.open()
                         self.serial_number = device.serial_number
                         device.set_raw_data_handler(self.handler)
                         
                         try:
                            device_mode = [0x0] * 33
                            device_mode[1] = 0x55
                            device_mode[2] = 0xAA
                            device_mode[3] = 0x20
                            device_mode[4] = 0x12

                            EPOC_ChangeMode = 1
                            
                            # EPOC 
                            if EPOC_ChangeMode == 1:
                                device_mode[5] = 0x00

                            # EPOC+ 128hz 16bit - MEMS off 
                            if EPOC_ChangeMode == 2:
                                device_mode[5] = 0x82                                

                            # EPOC+ 128hz 16bit - MEMS 32hz 16bit
                            if EPOC_ChangeMode == 3:
                                device_mode[5] = 0x86                                
                            
                            # EPOC+ 128hz 16bit - MEMS 64hz 16bit
                            if EPOC_ChangeMode == 4:
                                device_mode[5] = 0x8A
                            
                            # EPOC+ 128hz 16bit - MEMS 128hz 16bit
                            if EPOC_ChangeMode == 5:
                                device_mode[5] = 0x8E                                

                            # EPOC+ 256hz 16bit - MEMS off
                            if EPOC_ChangeMode == 6:
                                device_mode[5] = 0xE2                                
                                
                            # EPOC+ 256hz 16bit - MEMS 64hz 16bit
                            if EPOC_ChangeMode == 7:
                                device_mode[5] = 0xEA                                
                            
                            # EPOC+ 256hz 16bit - MEMS 128hz 16bit
                            if EPOC_ChangeMode == 8:
                                device_mode[5] = 0xEE
                                

                            report = device.find_output_reports()
                            report[0].set_raw_data(device_mode)
                            report[0].send()
                            print "Sending Packet!"
           
                         except Exception, exception:
                            print("Oops!",sys.exc_info()[0],"occured.")
                            print exception
                            print sys.exc_traceback.tb_lineno 
```

JulyMorning, 2017-10-13:
@warrenarea please, could you tell me if the current version of EPOC+ is supported? I'm considering to buy it and use for neurofeedback training, but it's restricted by the device by default, the metrics are limited and so on. Can I get what I need with the current libraries available?

warrenarea, 2017-10-13:
i think we're getting close to being able to use the Epoc+ using cykit or emokit, but
its not quite there yet. 

also, i don't even think the Xavier control panel displays the "high precision" data. 
maybe it does, but its definitely hard to tell because their auto-scale really messes up the visuals.

they recently came out with a new program, (that will cost you $100/mo) 
its called EmotivPRO.  I think you can definitely see a huge difference in the resolution... 
and at very least, the auto-scaling is fixed, 
It runs a service in the background... "EmotivCortexService" and I think has been improved for speed.... 
even though I notice it kind of jitters a little bit in the scrolling... but considering all of the different processing of things in the background, its no wonder. 

So to answer your question, from the current emokit libraries, they're not going to be much help
to you right now.

warrenarea, 2017-10-13:
It almost seems like, they never quite finished working on Xavier, before they switched to working on
EmotivPro.    Like they decided to redesign it for some reason...

Under the hood EmotivPRO looks completely different. lol 
Its like they really refocused their efforts toward security, because it has many different layers
of security.  Internally, I think it CRC's some of its own functions, uses SSL/TSL setup to 
communicate with emotiv, compares registry and files for authentication.  and on top of that,
i am pretty sure they ran it through an obfuscation program.... because the function/symbol names
are missing, and the functions seem like they loop around on themselves.

i think with Xavier, they were experimenting a little with polymorphic functions.. and then maybe just
used a couple tricks, to try to make it difficult for a decompiler to read.... (wasn't expecting that)

probably due to all of our activity on Emokit recently.... they have really switched gears to security. lol

warrenarea, 2017-10-13:
I think its very possible someone read that I had been working on Xavier, and decided to
step up their game.

JulyMorning, 2017-10-13:
@warrenarea hm, does some of you guys (the library's devs) have the current version of EPOC+ to experiment with? and, if not, is it the reason it's currently unsupported? I thought, maybe it's possible to establish a remote desktop connection, so you could try to hack it, but if the protection was increased, probably there's no way to do it, so it could be a waste of money for me, 800$ is 800$ and their basic SDK really sucks.

warrenarea, 2017-10-13:
I think the biggest shame of it all..... is that, the Emotiv developers.... have to work on security to keep the
raw data away from us..... and the Emokit consumers, have to work on security... to try to get that raw data back again.

So both sides are fighting over the data....  and the problem with that is,  we're so busy working on the
security.... that we're not able to do the actual neural research, and play with the headset we purchased.
all of our time is being drained into the security of the product.

i appreciate where Emotiv is coming from.... they want to be a profitable company, and I want them to
be profitable as well.... but I don't think "Emotiv as a service", is the answer... in order to make Emotiv a
huge global entity.... I think it should be opening itself up to the community, so we can see what 
amazing things can be created with it... when given open access to it.

I don't know about the rest of you, but the current Emotiv store (showcase), doesn't really do much
for me.

I also read recently that Disney is also going to be backing Emotiv as well, so you know they're getting
a lot of support.

warrenarea, 2017-10-13:
July:  
Currently, there is only a few developers here (as far as i know), who have the Epoc+
but I can not say how many of them are still working on it..... 

I thought we were making progress earlier, but then all activity seemed to drop off.
I've been working on it almost non-stop though. lol 

I have the EPOC+, Insight, and Epoc. to experiment with, 
The code I have above.... is the complete code to change the different Epoc+ modes via USB. 

July:
the encryption was increased only on the EmotivPRO application.... which only means its
more difficult to decrypt "that" particular application.... the other applications/libraries are
still fair game, and much easier to decipher.

reason why I haven't broken it... is because I have been on a "learning curve" to reverse engineer
applications. lol i know a lot about programming and a bit about assembly, but I still have 
much to learn about reverse engineering... I would be open to a remote desktop arrangement
if that would help...  I've gained a ton of insight already into the inner workings, but even then,
there are just so many functions to sift through. lol   just got to keep trying.... each time, i learn
a little more.

warrenarea, 2017-10-13:
their basic SDK, is still being updated consistently.... but the problem is, 
their lead developer(s) are probably so busy on the EmotivPRO, that they can
only update the SDK a little at a time.... and they are probably giving their
love to the C++ apps which they are more familiar with programmatically,
and not focusing as much on the Python examples.

warrenarea, 2017-10-13:
Also.... I have a repository called CyKIT which is a TCP streamer of the regular Epoc data.

Since then.... I've created a new engine for it... and a whole new interface that should open
up a lot of doors for us.... but before I released it....  I really wanted to get the Epoc+ precision
data nailed down.

Instead of tapping the sensors.... I've actually taken it a step further.... I spliced some needles 
to a speaker cable.... Attached the left speaker to 2 wires, and the right speaker to 1 wire.
Plugged the 2 needles into reference sensors..... and 1 needle into the target sensor.

and then I just play a low-amplitude frequency into my speaker..... and in that way, I can
watch the wave-form in the EEG data.   This is helpful... because i think it can tell me if I
have the masking correct or not. (else it will display the waveform wrong)

warrenarea, 2017-10-13:
Another issue that has slowed me down, is that.... the Epoc+ makes use of 64-bit floats,
so I'll have to be using Numpy in Python to make use of these double floats.

Problem is... there is quite a bit of floating point math in the more complicated functions.
So when I'm looking at the assembly..... I might think I'm looking at the data handling function...
but in fact, I'll be looking at the HAMMING/BLACKMAN or gaussian functions. lol
or just some other quality measurement function.

warrenarea, 2017-10-13:
or... because they loaded a huge library of encryption protocols.... i might end up looking at one of those.

JulyMorning, 2017-10-13:
@warrenarea I appreciate your comments a lot, thanks. Probably, the development of CyKIT/emokit-like project should be a bit more closed, to stop all the cold war with data, you know. They're watching this repos for sure.

I think I'll buy it as soon as you say you was able to unlock it, I trust you. But for now, it's too pricey toy considering the locked functional. Could you bump this thread if you have any luck with it?

warrenarea, 2017-10-13:
Well, I'll make sure to advertise the new repository for CyKIT,   i would keep checking that repository for
improvements.   It may be some time before I finally get it right though... 

I'm just a little worried Emotiv might not be very appreciative of my efforts though, but i've come this
far.... i hate to just turn back now. lol

JulyMorning, 2017-10-13:
hope you'll make it. (*crossed his fingers*)
I'm following your CyKIT project, but please, mention me here with @ when you think it's safe to buy it.

JulyMorning, 2017-10-25:
@warrenarea hello there :3 sorry to bother you, but… any news, ideas, opinions?