openyou/emokit #266: OSError: libhidapi-libusb.so.0: cannot open shared object file: No such file or directory

wangwuqi, 2017-07-03:
I install the emokit and  run the Usage example code, then it cant work
```
wuqi@wuqi-VirtualBox:~$ python test.py 
Initializing Emokit...
Initializing Reader Thread...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/emokit/emotiv.py", line 229, in run
    self.initialize_reader()
  File "/usr/local/lib/python2.7/dist-packages/emokit/emotiv.py", line 133, in initialize_reader
    self.reader = EmotivReader()
  File "/usr/local/lib/python2.7/dist-packages/emokit/reader.py", line 35, in __init__
    hidapi.hid_init()
  File "/usr/local/lib/python2.7/dist-packages/hidapi/hidapi.py", line 413, in hid_init
    __load_hidapi()
  File "/usr/local/lib/python2.7/dist-packages/hidapi/hidapi.py", line 164, in __load_hidapi
    __hidapi = CDLL(__libpath)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 362, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libhidapi-libusb.so.0: cannot open shared object file: No such file or directory
```

warrenarea, 2017-07-03:
i think its related to your libhidapi-libusb.so  file.

it looks like you're missing a file related to utilizing the usb.
also.... i'm not sure how things will work for you with virtualbox... 

you may have to try to hunt down those libhidapi-libusb.so files
or find out what library they're from...
i suggest a google search of 

```
OSError: libhidapi-libusb.so.0: cannot open shared object file: No such file or directory
```

warrenarea, 2017-07-03:
you may also want to investigate how virtualbox is communicating
in regards to the /hidapi/

i think i've had issues with the past using USB stuff in virtualbox, kind of
complicates matters.... i remember, you also needed to install an 
"extended virtualbox" addon as well.

wangwuqi, 2017-07-04:
I install emokit on windows10, and run the Usage example code, and the reader just 'stopping, stopped and restart', 
```
Reader stopping...
Reader stopped...
Restarting Reader
Initializing Reader Thread...
```

wangwuqi, 2017-07-04:
well, I find the similar issue 'Reader Thread Keeps Restarting #218'

xribene, 2017-07-04:
I had the same issue, until i tried in ubuntu ( not in virtual box) and all
hidapi and reader restarting problems were solved

warrenarea, 2017-07-07:
I think I'll eventually have to setup linux so I can assist with updating the repository, 
it would be nice to get it working in virtualbox too.. because then it could be an easy switch
back and forth from Windows.

At this time though, it doesn't make sense for me to commit my time and mental resources to
setting up the OS, and tweaking the scripts settings to get it to work.  Makes more sense to continue
trying to decipher the workings of these devices, first figuring out how they work before spreading
myself over two architectures.

GabrielBioML, 2017-08-08:
I had the same problem. I fixed it by changing a line hidapi.py. It's dirty but it works. at line 157 you have the function find_library but it doesn't return the complete path only the name of the library. What I did was to had the rest of the path to the shared library at line 164:
        # Search for the hidapi library.
        __libpath = find_library('hidapi') or\
                    find_library('hidapi-libusb') or\
                    find_library('hidapi-hidraw')
	print(__libpath)
        if __libpath is None:
            raise RuntimeError('Could not find the hidapi shared library.')

        # Load the hidapi library.
        __hidapi = CDLL(r"/usr/local/lib/{}".format(__libpath))  #this is 164 where I added the rest of the path
        assert __hidapi is not None

hope it helps

warrenarea, 2017-08-08:
Also. Epoc+ uses the hidapi product name of 'EEG Signals'