openyou/emokit #267: battery_values KeyError

wangwuqi, 2017-07-04:
the running result is as follows, and it generate a csv file 'emotiv_values_2017-07-04 22-43-55.655133.csv', but there are only 4 row data,what should i do

```
Initializing Emokit...
Initializing Reader Thread...
Reader detected serial number: UD201605030023AB
Initializing Writer Thread(s)...
Initializing Crypto Thread...
Initializing Output Thread...
EmotivCrypto: Generating new AES cipher.
EmotivCrypto: Serial Number - UD201605030023AB
EmotivCrypto: Generated Crypto Key from Serial Number...
   Serial Number - UD201605030023AB | is research - True 
   AES KEY - ['B', '\x00', 'A', 'H', 'B', '\x00', 'A', 'T', '3', '\x10', '2', 'B', '3', '\x00', '2', 'P']

Emokit - v0.0.8 SN: UD201605030023AB  Old Model: False
+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   |    0     |    0     |  Nothing   |  Nothing   |
|   FC5  |    0     |    0     |  Nothing   |  Nothing   |
|   AF3  |    0     |    0     |  Nothing   |  Nothing   |
|   F7   |    0     |    0     |  Nothing   |  Nothing   |
|   T7   |    0     |    0     |  Nothing   |  Nothing   |
|   P7   |    0     |    0     |  Nothing   |  Nothing   |
|   O1   |    0     |    0     |  Nothing   |  Nothing   |
|   O2   |    0     |    0     |  Nothing   |  Nothing   |
|   P8   |    0     |    0     |  Nothing   |  Nothing   |
|   T8   |    0     |    0     |  Nothing   |  Nothing   |
|   F8   |    0     |    0     |  Nothing   |  Nothing   |
|   AF4  |    0     |    0     |  Nothing   |  Nothing   |
|   FC6  |    0     |    0     |  Nothing   |  Nothing   |
|   F4   |    0     |    0     |  Nothing   |  Nothing   |
|   X    |    0     |   N/A    |    N/A     |    N/A     |
|   Y    |    0     |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    0     |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:    2     |  Packets Processed:    0   |
|   Sampling Rate:    2     |        Crypto Rate:    0   |
+========================================================+

Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Python35\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  File "C:\Python35\lib\threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Python35\lib\site-packages\emokit\emotiv.py", line 298, in run
    new_packet = EmotivPacket(decrypted_task.data, timestamp=decrypted_task.timestamp)
  File "C:\Python35\lib\site-packages\emokit\packet.py", line 39, in __init__
    self.battery = battery_values[str(self.counter)]
KeyError: '206'
```

xribene, 2017-07-04:
in ubuntu i  get the similar results when 
a) emotiv is not close to the usb dongle so there is not good pairing. When i move close and restart the process then everything is fixed. 
b) When i am running a python script that uses emotiv if i terminate the program by closing the terminal instead of the right way (Ctrl-C , headset.stop()) , then the next time i run the script, i get the same KeyError: '206' . To fix this i have to remove the dongle and restart the process again.
I dont know if it is the same in windows .hope it helps you

wangwuqi, 2017-07-04:
I run this on win10
what's your headset manufacturing date, is my headset too new to crack?

xribene, 2017-07-04:
I don't know exactly the date. I am the third owner . but for sure it is
not emotiv+ . I guess the date is around 2012. I am not position to answer
to your question, i have not follow all the last updates. But if you search
in other issues, there are some other guys here that can help you.

wangwuqi, 2017-07-04:
would you mind if i can  have your wechat or twitter connection . I want to have some related discussion with you.

xribene, 2017-07-04:
I dont have account in any of these. Maybe you could send me pm ?

pyni, 2019-07-02:
Hi, i meet the same problem.
Have you solved it ?