openyou/emokit #269: Gyro

aro280994, 2017-07-13:
I've been worked with Python, but whe I ran example.py code, the gyro signals are 0 all the time. 
How can I solve this problem? 

Help! Thank you.

warrenarea, 2017-07-13:
what headset model are you using?

aro280994, 2017-07-13:
I'm using EMOTIV EPOC+

warrenarea, 2017-07-13:
yes, that will explain why its not reading the values.

The Epoc+ version is being worked on... I have managed to get the gyro working for the Epoc+

#264 
Currently the Epoc+ in 256hz mode decrypts with key model 6
In Epoc-mode it will be key model 4

#265 
This is code that, if you have it detect "EPOC+" product name... you can get it to
"change modes", while the headset is turned on, and plugged into USB.

The first byte, is the counter, it will go from 0-255

If the 2nd byte is 16, then thats the EEG data packet.

If you iterate through the raw data variable... you will see 14 "127" values (if the headset is not worn)
these are signal quality values... when its worn, it will change either slightly below or above 127.

Next to every 127 value, is the raw data value... you won't get any floating point math from it right
now, but you should be able to use it for basic EEG.

after 14 signal quality+data values, you see 0 0
then another 14 signal quality+data values

If the 2nd byte is 32, that is the MEMS (gyro data)

I used code in another language, but you should be able to make use of it.

```
    if ($4 < 255 && $4 > 128) { inc %X $calc((255 - $4) * 5) } 
    if ($4 < 255 && $4 < 128) { dec %X $calc($4 * 5) }

    if ($6 < 255 && $6 > 128) { dec %Y $calc((255 - $6) * 5) } 
    if ($6 < 255 && $6 < 128) { inc %Y $calc($6 * 5) }

    if ($8 < 255 && $8 > 128) { dec %size $calc((255 - $8) * 1) } 
    if ($8 < 255 && $8 < 128) { inc %size $calc($8 * 1) }
```
so the 4th byte was to move the X axis
the 6th byte was for Y axis
and 8th byte was Z axis

To go left, it subtracts the motion amount from 255.
To go right, it starts at 0 and adds the motion amount.

I will probably throw some code up later on.

With my CyKit version, i've just been streaming the raw data, and working with it elsewhere.