openyou/emokit #270: Unittests fails - EmotivPacket not defined?

jnvandermeer, 2017-07-17:
Hi everyone,

First off thanks for keeping this repo updated. For me the emotiv stuff is really a non-option if I have to rely on closed-sourced software just to access the raw data.

For me the python2 -m pytest tests/ fails. I tried finding and grepping anything like "EmotivPacket", but.. this class is not defined anywhere. I did find EmotivNewPacket, though.

============================= test session starts ==============================
platform linux2 -- Python 2.7.13, pytest-3.1.3, py-1.4.34, pluggy-0.4.0
rootdir: /home/johan/builds/emokit/python, inifile:
collected 4 items / 1 errors 

==================================== ERRORS ====================================
____________________ ERROR collecting tests/test_packet.py _____________________
ImportError while importing test module '/home/johan/builds/emokit/python/tests/test_packet.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
tests/test_packet.py:3: in <module>
    from emokit.emotiv import EmotivPacket
E   ImportError: cannot import name EmotivPacket
!!!!!!!!!!!!!!!!!!! Interrupted: 1 errors during collection !!!!!!!!!!!!!!!!!!!!
=========================== 1 error in 0.12 seconds ============================

The required info:
sys.version_info(major=2, minor=7, micro=13, releaselevel='final', serial=0)
Linux
-------------------------
usage_page, 0
product_id, 8455
interface_number, 0
manufacturer_string, None
vendor_id, 16700
release_number, 260
serial_number, None
usage, 0
path, 0001:002d:00
product_string, None
-------------------------
usage_page, 0
product_id, 20002
interface_number, 0
manufacturer_string, None
vendor_id, 1121
release_number, 256
serial_number, None
usage, 0
path, 0001:002c:00
product_string, None
************************************************************
! Please include this information if you open a new issue. !
************************************************************

-- I have to add that I try to connect the EPOC+ Headset (3B9ADC95), and it connects, but whenever I try to run examply.py, I get a Device not found exception error. Separate issue, probably, though.

-- I also have to add my system is Arch

Tomorrow I will try on my Windows partition to see how I fare there.

Thanks
Johan

ivankras, 2017-08-24:
The only class on the file is Emotiv, not EmotivPacket. Thus, I renamed it on test_packet and the problem disappeared. I'm getting another issues, but I don't think they're related to that

jameel263, 2017-10-01:
It looks like the class `EmotivPacket` was "replaced" in this commit `6f93df9994dad30e5e9e894432258dda510f141f` but the tests weren't updated accordingly.