openyou/emokit #272: Large values

ShaneHudson, 2017-09-03:
I'm new to emokit, and am wanting to use the raw data from an EPOC+. It doesn't need to be 100% accurate, but I want it to be useable for detecting differences to show readings while listening to different types of music.

As you can see from the screenshot, the values are coming back much larger than expected. Am I interpreting them wrong? Or is there an issue?

[image]

Vsevalot, 2017-09-06:
If I understood datasheet right, epoc+ has resolution 14 bit, with LSB = 0.51μV, so to get voltage value you should subtract 4096 from your data and multiply result to 0.51μV

Epoc + datasheet: https://www.emotiv.com/files/Emotiv-EPOC-Product-Sheet-2014.pdf

ShaneHudson, 2017-09-06:
Ah brilliant thank you!

closed 2017-09-06 by ShaneHudson