openyou/emokit #279: Looking for yet another project lead

qdot, 2017-11-09:
The last commit for this repo was in April, and it looks like bschumacher is no longer on github. I think we may need a new project lead. Anyone interested?

warrenarea, 2017-11-09:
yes, i was reviewing something Bill said about the Epoc+ on another thread.
 i think he actually was really really close to nailing down how we process Epoc+ and converting
to 64-bit double float.

qdot, i started my own version CyKIT, which streams the data via TCP, 
but I upgraded that yesterday with CyKITv2 
https://github.com/CymatiCorp/CyKITv2

What makes this special, is that it gets around the XSS cross-browser limitation of
accessing the browser locally by giving it a secure handshake, and allows data to be 
transmitted from Python directly into the browser. 

Meaning we can now access the data directly from the browser... which opens the door for
our Emotiv projects to have access to not only javascript, but lets us use the browser as an
interface for rendering the data.

I personally feel that is the direction we should be taking this.... because it seems like we have
used pygame for rendering data in the past... but as many strengths Python has, I don't think
rendering graphics is really one of them.   and I don't think I've ever seen anyone really feel
encouraged to really interact or do anything with pygame or its interface.

By turning emokit into more of a 'socket client' instead of a stand-alone application with examples,
We can use the "same" HTML/JS examples for "all" of the operating systems, because they all typically
make use of Chrome or Firefox, or make use of HTML5 standards.

and I think in this manner, we could potentially have a lot more examples for people... especially,
if I know a lot of people out there, are more comfortable or fluent with Javascript. They can
just dig in and start creating. 

Since I work primarily with Windows, i will need assistance with setting up a linux port, of
the architecture i set up.

So here is my idea.... I would like to be the project lead, and help engineer this new architecture
for emokit, but if it is possible, I would like you to select someone else, who wouldn't mind 
taking care of the Linux port of what I have set up.  Which should work smoothly because, I don't
think it was ever a good arrangement having both Windows+Linux+OSx support in one file, 
think it would be cleaner to have them separate, and would help maintain it a little easier.

CyKITv2 proves that it is do-able, and the browser actually scrolls even faster than Emotiv's 
programs... So speed/performance does not seem to be an issue here.   The architecture is set
up as well, to provide multiple support for all 3 different headsets (Epoc, Epoc+, Insight)
though more research needs to be done sorting out the floating point math. 

What are your thoughts?

qdot, 2017-11-09:
Hah. Small world. I'm working on the exact same type of project, just with far, far different devices.

One of the many reasons I can't pick this back up is because I'm currently working on https://buttplug.io (NSFW), which is the same idea as CyKITv2, except for sex toys. I've been working in sex tech for over a decade, my jaunt into EEGs was part of health tech work I just did for a couple of years and am not really involved in much anymore. Basically, we're abstracting bluetooth/usb/serial/whatever transports and folding it into a simple protocol and client/server system that anyone who can build a JSON packet can hopefully use.

For windows, we went with hosting a local websocket server in C# that provides the bridge to bluetooth/usb/serial. On Mac/Linux/Android, we just went with node.js with WebBluetooth currently because it "just works" in chrome and other browsers are looking at implementing it.

Back when I was working on emokit, I was kinda trying to nudge it in that direction too, which was partially why I made OSC output. There's not so much a need for library embedding as there is for just a quick, simple way to get the data out of the device to do something with it, which it looks like you're trying to do.

So, tell ya what. Gimme a few days to figure out where things are with emokit and your project, and I'll get back to you on this. I haven't seen my emotiv headset in a couple of years, I'm assuming it's in a random box somewhere, so I don't expect I'll be actively testing anything, but I can at least spend a bit of time helping on whatever next steps are here.

warrenarea, 2017-11-09:
ah yes, i actually did consider the Node.js route... had it all installed and ready to go,
but my whole theme and concept so far, has been to try to optimize and remove as
many "moving parts" as i can. 

In fact, now that I recall, the node.js method didn't even work properly, apparently the
windows version wasn't quite up to speed.

(For example, took out the need for gevent, and greenlets... and just do straight up
threading... which Bill also arrived at the same solution.) There is actually an even faster
method of threading, if needed.

 I got lucky though and stumbled across a Python web server, that does the handshaking.
In fact... I figured even if CyKIT wasn't successful, I was just going to offer up the 
server platform as a github project.... the original owner had it all separated into a bunch
of files, and i managed to unify them into a nice tidy package.  plus it was in
chinese or japanese.... i think his was probably extracted from the Google's pywebserver 
which is also really bloated. 

Firefox and Chrome both work... but I have to say, Chrome is much faster/responsive, especially
when it  comes to the debugging window.... Firefox isn't very debug friendly, and will likely just 
crash if you log too much data.

i appreciate the work bill has done, and his debugging examples are pretty brilliant, where
everything does seem very modular, but i also kind of think there is too much going on in the
repository, like key cracking and mask testing... i would likely do a ton of pruning, getting it down
to its bare essentials.... there is so much there, I would almost hate to remove anything in case
someone might find something useful.... but at the same time, it needs a spring cleaning,
a nicer faq, and pictures!

warrenarea, 2017-11-13:
qdot, have you had any time to review the project?

qdot, 2017-11-18:
Running a little behind, gonna try to get to this soon.

warrenarea, 2017-11-18:
I've been making some interesting discoveries about how the Epoc+ uses its data.
Think you'll be around long enough you might be able to assist me in making some sense of the math,
think i'm pretty darn close.

warrenarea, 2017-11-20:
https://discord.gg/gTYNWc7
If anyone would like to assist me in the Epoc+ data
I have a Discord chat server setup here.

You can either download the discord chat messenger,
or just click the link to use the browser interface.

Sitch, 2017-11-26:
Have you guys considered NPL? 
https://github.com/LiXizhi/NPLRuntime/wiki/WhatIsNPL

Excels at multicore/distributed visualization (2d/3d), neural nets, etc.

qdot, 2017-11-26:
Ok, I give up, was really trying to put some time aside but I don't have resources to deal with this.

More than anything, I'd just like to see documentation of the protocols. I'm working on this for other projects right now using gitbook format. Having decent documentation means people aren't necessarily bound to one repo's vision of how the hardware should be used, though they'd also need the ability to implement things themselves.

That said, I really can't help on that. I haven't worked with eegs in years, and I am far too busy with my other projects to do much meaningful here.

So, @warrenarea, the one thing I can't really discern from your comments is, are you looking to combine cykit and emokit, or just strip down emokit but leave it as is? The problem with making emokit rely on another framework like cykit is that it's always been a standalone thing, albeit a not very clean standalone thing. I'd much rather someone else start something from scratch than change what emokit is yet again, especially since Bill did the packaging work on pypi and we don't have those keys. :|

warrenarea, 2017-11-27:
Well, I was originally thinking it would be nice to integrate the browser core
that I had been using with CyKITv2. Because even the examples in emotiv usually
made use of "PyGAME" to do some rendering. 

However if it is your intention just to have a generic core that people can try to plug-in
to their projects, I think I could do that.

What I really really want to do though with emokit, is strip it down to the bare necessities,
because currently... the main folder is riddled with "testing" scripts that were meant to figure
out the encoding process. 

I tried to install emokit the other day, just to try to run it, and i spent quite some time just 
downloading all of the dependencies for it... currently i think it really is bloated in that area.

my goal would be this:

A. Clean out the python folder of unnecessary testing scripts (almost all of them)
B. Remove pyTest /tests folder
C. Write a more thorough FAQ
D. Condense the files (within reason)
E. Separate Windows from Linux and place in different folders.
F. Update code to new standards, for:  epoc, epoc+ and insight.
G. Write a basic example for usage.
H. Update readme.md so it looks cleaner and nicer, and incorporate pictures
    demonstrating usage.

After I spoke with you the other day, I managed to figure out how the Epoc+ decodes
its data.  Its still not perfect, the decimal value is accurate, but I am currently getting up
to 6 floating digits of precision to match the EDK.  I figure for the time being, that is 
close enough, and can at least point others in the right direction.

Also, the emotivPRO (their new software) is even more accurate with up to 10 digits of
floating point precision... which should be a lot easier to solve, now that i know their
protocol for it.

having all of the headsets at my disposal, i think it puts me in a unique position to help
move this project forward.  plus, i really would like to take care of some of these goals of
cleaning up this repository, and getting it back on track.

and with this new data encoding under my belt, i think i can bring it up to speed.

warrenarea, 2017-11-27:
Part of the problem i had with pygame is that i don't feel people felt really comfortable using that
interface to really "do" anything with their data.

that's really why i pushed CyKIT in the direction from being just a TCP data streamer,
in the direction of using the browser as an interface, is because i think the browser is
more accessible to developers, especially with javascript in mind.

another thing i kind of notice that might occur sometimes, is that people tend to get
distracted by the graphics.  once a project reaches a developmental 'epoch' of displaying
graphics, i think there is kind of a slow-down in productivity. 

why in the direction of emokit, it might be best just to do what we had been doing....
and just displaying a standard text output... as well as the basic CSV writer. 

I have currently just finished a CSV writer in CyKITv2 and am now working on various 
data outputs, that people might want to use. 

Another thing I can bring to the emokit project... is that I sorted out "sending" data to
to the Epoc+ headset, to get it to change what mode it is in.  Something else that can be
incorporated.

warrenarea, 2017-11-27:
kyle, tell you what, put me in charge... and if you don't like what you see, 
you can always revert the changes back, or nudge me in a direction you see
more fit for the project. 

i am pretty easy to work with, and i think i am fairly reasonable.
what do you think?

qdot, 2017-11-27:
'k, you're added to the group with write perms. I mean, not exactly like I did a ton of vetting with Bill either. 

I'm on board with the pygame ungluing, I just wanted to make sure that people have a core library to access if they want to do their own thing too. However, I have zero visibility into whether anyone that uses this library actually WANTS that (I'm basically an absentee owner here), so I'm gonna try to trust you on the direction of things for now.

qdot, 2017-11-27:
Closing issue for the time being since at least someone that cares has write perms now. We can break discussion of the project course out into more issues if needed, but most likely I'm probably just gonna disappear again. Feel free to tag me specifically if I'm needed somewhere.

closed 2017-11-27 by qdot

warrenarea, 2017-11-27:
Okay, thank you for the support.

As I mentioned earlier, I have created a discord channel for this repository, so
that can assist in getting the feedback on what people need.

I think you will be pleased with how i set things up.