openyou/emokit #280: Epoc+ and horrific terms

coristus, 2017-11-13:
Hi all, I bought an emotiv epoc+ after a chat with an employee months ago, telling me it would work on linux. Saved money for months and finally bought one. Omly to find oit that they encrypt their datastreams. Through the sdk you can login in (online) and get a session key.
I advice everybody NOT TO BUT EMOTIV. Just read the terms of service and you'll understand.
According to the tech support there is quite some IP involved in how they present raw data????? That is one bs argument if I ever heard one.
Is there anyone with similar experience... or someone interest in a brand new epoc+?
STAY AWAY FROM EMOTIV.

warrenarea, 2017-11-13:
If it was a recent purchase, you might see if you can return it.
Might contact their help-desk and get more information. 

I am sorry to hear you are having an unpleasant experience. 

It is quite a large purchase to make... your co-worker was likely speaking of the Epoc version
that has been decrypted for the raw data.

You can also get the raw data by using their EDK.dll  but like you said, you have to purchase a session key.

i do agree, i don't like their "pay as a service" procedure regarding raw data.... i mean,
it is "your" data, and i think that whole procedure limits the user from actually "working" with the
data and performing actual research. 

I was told recently they are working to set up an "unlimited" access to raw data. 

For $30/month they offer 30 accesses to the data via a session key...
i think "that" is ridiculous, because each session is supposed to last 30 minutes.... but if you have
an error in your program.... that 1 run failed run, counts as an access/session. 
Hell, i can have 30 errors in one programming session alone.

For $79/mo you can use their "Free Trial" of EmotivPRO for 14 days.... what you can do is...
find the Ecvn.bin file, and get your license key (or acquire the license key from the website)
and you can plug that into the Python Community-SDK.

That will give you "unlimited" data access, for those 14 days.

You can then cancel your subscription, or purchase it for a month.... at least then, you can have
access to the raw data for a while... and decide if you can do something cool with it.

coristus, 2017-11-13:
Hi warren,
Thanks for your reply. The thing is I don't have windows anywhere. So.. it won't be resolved unfortunately. And even if I were to install .. I couldn't return it anymore I think. I'm Now working on a refund. But I will have spent 250 dollars on shipping and taxes, AND have a 25% restocking fee. I'd be happy to sell you mine :) brand new. Worn once, nice extra 100 felts. Haha. 
I know exactly what to do with raw data. I use neural nets to classify them :). That's it's sole purpose. For a thesis. I am going to switch to a device that doesn't extort me. OpenBCI . Once I get my money back that is. 
But I'll still have lost 450$ on shipping and restocking. Would they have notified me of this detail I woukd never have bought one. The friendly chat agent months ago told me it would work on Linux. I thought that meant exactly that..
So if you want to do something cool with it pm me.

warrenarea, 2017-11-13:
coristus, you should still be able to get it to work on Linux...

the community-sdk has a few different languages you can use.... Python in particular,
which... you should be able to use with Linux.

Eventually we might come up with a Linux solution in Python as well for raw data... so you can't
give up hope on that either.

Perhaps it won't be ready for you by thesis time, but... you should still be able to use a linux
setup with their Python Community-SDK (on github) https://github.com/Emotiv/community-sdk

and just sign up for the free trial of EmotivPRO to get that license key i was talking about.

warrenarea, 2017-11-13:
also consider.... that with OpenBCI, you will be paying a bit more, and won't get as many sensor
contacts to use.

also.... Epoc+ is special, in that, you are getting .13uf (micro volts) which is a bit higher sensitivity.

I really think you should keep your Epoc+, yes, it might not be exactly what you had in mind, but
at least this way, you won't take any loss by keeping it...

coristus, 2017-11-13:
Hi warren,
Thanks again for replying. The sensitivity issue is a very good argument... 8 sensors on the openbci is enough for my purpose.. but sensitivity is ... probably more important. I still have a few month before I start working on my thesis, I decided to buy one now to familiarize myself with it and to prevent any (nasty) surprises from killing my project. This seems to be one of those.. 

Will it then be possible to access raw data without a subscription, or am i reading it wrong?

Also, python is my preferred language, I build my models in python/keras/tensor/theano. So anything python I can do.

So keep my epoc+ ... i'll give that some serious thought... Your argument is good.
regards,
Cory

warrenarea, 2017-11-13:
If you subscribe to the Free Trial, you will have 14 days of unlimited raw access to the data.

Then you can always cancel before its due.

In theory, the Python version should work for you, I have tested it myself this past month.

I think its hardest just making use of the data... unless you have an algorithm that you can run
it through, like they do for cognitive detections. I haven't got to experiment much with that. Been 
so hung up on decrypting, haven't got to enjoy the raw data.  

If you know it can be decrypted though.... I think you should find comfort in the long-term, that
you will be able to use your device later on (down the road).   and you obviously bought it with 
some intention in mind, and intention to keep it.

I think you have buyers remorse right now, lol it will pass, it often happens on large purchases,
when i bought my devices i had it a little bit, kind of questioning... did i make the right choice? 
so far, i am happy with my purchases.... as I have much to do with them... so just push your doubts
to the side, and focus on your projects, and at very least discover what it is capable of, before 
you try finding it a new home.

tahesse, 2017-11-19:
Hi, I was trapped too. Though I want to spend my free time to ultimately get the raw data from the Emotiv EPOC+. Before I'll start from scratch, are there links that point me in the right direction? I'll check the resources of this repository to start. :-)

I initially bought the Emotiv EPOC+ in order to follow along my BCI journey as I just have little experience (roughly one year). Sadly, I was stupid enough not to see their _retarded_ licensing. :1st_place_medal:

warrenarea, 2017-11-20:
https://discord.gg/gTYNWc7
If anyone would like to assist me in the Epoc+ data
I have a Discord chat server setup here.

You can either download the discord chat messenger,
or just click the link to use the browser interface.

coristus, 2017-11-30:
I will this weekend. Bit busy with research deadlines and exams atm

sidharthramesh, 2017-12-26:
@coristus have you tried the CyKITv2?