openyou/emokit #282: RuntimeError: Could not find the hidapi shared library.

SHiLLySiT, 2017-12-09:
When attempting to run python/example.py I get the following error:

```
Initializing Emokit...
Initializing Reader Thread...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/alarioza/Desktop/emokit/python/emokit/emotiv.py", line 234, in run
    self.initialize_reader()
  File "/home/alarioza/Desktop/emokit/python/emokit/emotiv.py", line 137, in initialize_reader
    self.reader = EmotivReader()
  File "/home/alarioza/Desktop/emokit/python/emokit/reader.py", line 35, in __init__
    hidapi.hid_init()
  File "/usr/local/lib/python2.7/dist-packages/hidapi/hidapi.py", line 417, in hid_init
    __load_hidapi()
  File "/usr/local/lib/python2.7/dist-packages/hidapi/hidapi.py", line 162, in __load_hidapi
    raise RuntimeError('Could not find the hidapi shared library.')
RuntimeError: Could not find the hidapi shared library.
```
But the hidapi appears to be installed correctly:
```
user:~/Desktop/emokit/python$ ls /usr/local/lib/python2.7/dist-packages
builtins                        hidraw.so                queue
copyreg                         hid.so                   reprlib
_dummy_thread                   html                     socketserver
emokit                          http                     _thread
emokit-0.0.8-py2.7.egg-info     libfuturize              tkinter
future                          libpasteurize            winreg
future-0.16.0-py2.7.egg-info    _markupbase              xmlrpc
hidapi                          past
hidapi-0.7.99.post21.dist-info  pyhidapi-1.0.0.egg-info
```
I'm not sure if this was a side affect of having to manually install [pyhidapi](https://github.com/NF6X/pyhidapi) from its git repo. As I was having issues installing emokit with `pip install emokit` because pyhidapi did not appear to be in the PyPi:
```
user:~/Desktop/hidapi/hidtest$ sudo pip install emokit
The directory '/home/user/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/user/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting emokit
  Downloading emokit-0.0.8.tar.gz
Requirement already satisfied: pycrypto==2.6.1 in /usr/lib/python2.7/dist-packages (from emokit)
Collecting future (from emokit)
  Downloading future-0.16.0.tar.gz (824kB)
    100% |████████████████████████████████| 829kB 962kB/s 
Collecting pyhidapi (from emokit)
  Could not find a version that satisfies the requirement pyhidapi (from emokit) (from versions: )
No matching distribution found for pyhidapi (from emokit)
```
Any ideas?

warrenarea, 2017-12-10:
i'm not sure about pyhidapi, 
we always used pywinusb, but that is probably just for Windows.

will have to sort this out later.  i personally could never get everything completely
working for it, even after i got all of the matching dependencies running.

i will be working on the repository later on, so will sort something out, and make
it easier to acquire the dependencies.

SHiLLySiT, 2017-12-11:
@warrenarea I will be looking forward to that!

JehuSilva, 2020-01-14:
Hello SHiLLySiT, I have the same issue. Did you solve it?

SHiLLySiT, 2020-01-14:
@JehuSilva Unfortunately no. I haven't worked on the project since 2017!

warrenarea, 2020-01-14:
@JehuSilva have you tried checking out the CyKIT project?   
i think some people have got it working for linux.

https://github.com/CymatiCorp/CyKit

JehuSilva, 2020-01-14:
> @JehuSilva have you tried checking out the CyKIT project?
> i think some people have got it working for linux.
> 
> https://github.com/CymatiCorp/CyKit

I have already seen the project somewhere, but I have not used it. Since I haven't solved the same problem with emokit, I'm going to look for it and try it, thanks for the suggestion @warrenarea .

JehuSilva, 2020-01-14:
> @JehuSilva Unfortunately no. I haven't worked on the project since 2017!

Thank you too @SHiLLySiT